UnipiTechnology/evok #99: can not install evok on fresh install of raspberry os or rassbian buster

lukasx102, 2022-08-31:
### Prerequisites

* [ ] Are you running the latest Master or the latest release version?
* [ ] Did you try to reinstall EVOK?
* [ ] Did you perform a cursory search on the [forum] and [google]?

### Log files needed

Include the output of the following commands on your Neuron/UniPi device:

```
sudo su
cat /var/log/evok.access.log
ps -fax
cat /var/log/evok.log
```

### Description

[Description of the bug or feature]
Hello,

please can you confirm that evok is possible on fresh installation of Raspberry OS?

I faced with a lot of issues during installation :

ERROR: The certificate of ‘repo.unipi.technology’ is not trusted.
ERROR: The certificate of ‘repo.unipi.technology’ has expired.
gpg: no valid OpenPGP data found.
root@raspberrypi:/home/pi# 
root@raspberrypi:/home/pi# apt-get install evok
Reading package lists... Done
Building dependency tree       
Reading state information... Done
W: Target Packages (main/binary-armhf/Packages) is configured multiple times in /etc/apt/sources.list.d/unipi.list:1 and /etc/apt/sources.list.d/unipi.list:2
W: Target Packages (main/binary-all/Packages) is configured multiple times in /etc/apt/sources.list.d/unipi.list:1 and /etc/apt/sources.list.d/unipi.list:2
W: Target Translations (main/i18n/Translation-en_US) is configured multiple times in /etc/apt/sources.list.d/unipi.list:1 and /etc/apt/sources.list.d/unipi.list:2
W: Target Translations (main/i18n/Translation-en) is configured multiple times in /etc/apt/sources.list.d/unipi.list:1 and /etc/apt/sources.list.d/unipi.list:2
W: Target Translations (main/i18n/Translation-en_US.UTF-8) is configured multiple times in /etc/apt/sources.list.d/unipi.list:1 and /etc/apt/sources.list.d/unipi.list:2
W: Target Packages (main/binary-armhf/Packages) is configured multiple times in /etc/apt/sources.list.d/unipi.list:1 and /etc/apt/sources.list.d/unipi.list:3
W: Target Packages (main/binary-all/Packages) is configured multiple times in /etc/apt/sources.list.d/unipi.list:1 and /etc/apt/sources.list.d/unipi.list:3
W: Target Translations (main/i18n/Translation-en_US) is configured multiple times in /etc/apt/sources.list.d/unipi.list:1 and /etc/apt/sources.list.d/unipi.list:3
W: Target Translations (main/i18n/Translation-en) is configured multiple times in /etc/apt/sources.list.d/unipi.list:1 and /etc/apt/sources.list.d/unipi.list:3
W: Target Translations (main/i18n/Translation-en_US.UTF-8) is configured multiple times in /etc/apt/sources.list.d/unipi.list:1 and /etc/apt/sources.list.d/unipi.list:3
E: Unable to locate package evok

I do not want to use unipi node red image, because I can not configure there hidden wifi..

please help me
thanks
### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expected to happen]

**Actual behavior:** [What actually happened]

[forum]:https://forum.unipi.technology/category/4/official-evok-api
[google]:http://www.google.com/

qcraftd, 2023-04-09:
64 bit ?

kratochvil01, 2024-06-20:
Evok v3.0.1 is released. (for debian bookworm)

closed 2024-06-20 by kratochvil01