UnipiTechnology/evok #102: Can't install EVOK for Unipi 1.1 under Raspian Bookworm 64 Bit

JR-Home, 2023-11-02:
Description:
When I try to install Evok 2.4.34. on a clean Raspbian Bookworm with 64 Bit, I got the following error Message:

########################
## Installing EVOK... ##
########################
cp: cannot overwrite non-directory '/boot/overlays' with directory 'boot/overlays'
####################################
## Using kernel newer than 3.18.5 ##
####################################
############################
# Using device tree kernel #
############################
Hit:1 http://deb.debian.org/debian bookworm InRelease
Hit:2 http://deb.debian.org/debian-security bookworm-security InRelease        
Hit:3 http://archive.raspberrypi.com/debian bookworm InRelease                 
Hit:4 http://deb.debian.org/debian bookworm-updates InRelease                  
Hit:5 https://deb.nodesource.com/node_18.x nodistro InRelease                  
Reading package lists... Done                                
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Package python-pip is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source
However the following packages replace it:
  python3-pip

Package python-dev is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source
However the following packages replace it:
  python-dev-is-python3

E: Unable to locate package python-ow
E: Package 'python-pip' has no installation candidate
E: Package 'python-dev' has no installation candidate
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done

python3-distutils is already the newest version (3.11.2-3).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
error: externally-managed-environment

* This environment is externally managed
    To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.
    
    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.
    
    For more information visit http://rptl.io/venv

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.

### Steps to Reproduce

1. wget https://github.com/UniPiTechnology/evok/archive/2.4.34.zip
2. Unzip...
3. bash install-evok.sh
4. nginx web server is not reachable

**Expected behavior:** 
EVOK will be installed, and nginx web server is reachable
**Actual behavior:** 
nginx webserver is not reachable

martytriska, 2023-11-03:
Hello, @JR-Home ,

unfortunately, now, Evok does not support Raspberrypi OS based on Debian 12 due to broken package dependencies. We are working hard to fix it :-). Please use older Debian 11 instead.

Moreover, from the next release, Evok will be completely rewritten in Python 3.

rudyvan, 2023-11-08:
Dear @martytriska,   any ETA on the python3 release you mentioned?
Thanks!

martytriska, 2023-11-20:
Hello there, the release is planned until the end of the year. Now, we are in the phase of testing and doc creation.

PEPITO82, 2023-11-20:
I would also be happy to help. Have a spare UniPi 1.1 which I would like to upgrade to a RPI 5 setup.

rudyvan, 2023-11-20:
same with me here. have a RPI5 with UniPi lite that is ready for experiments on python 3.12.0

PEPITO82, 2023-12-18:
Any update @martytriska ? End of year is close.

PEPITO82, 2024-01-11:
@martytriska Happy new year. Any update?

kratochvil01, 2024-04-09:
Hello, you can try [beta evok](https://github.com/UniPiTechnology/evok/tree/dev-bookworm).

PEPITO82, 2024-04-09:
@kratochvil01 Is it possible to install new evok beta on HAOS (home assistant) image that can be selected from the Raspberry Pi Imager?

kratochvil01, 2024-04-09:
The installation procedure assumes a Debian 12 Bookworm system. If HAOS is built on this system, the installation procedure should be the same as when [installing on Raspberry Pi OS](https://evok.readthedocs.io/en/latest/installation/#raspberrypi-os).

cewbdex, 2024-04-10:
Hello @PEPITO82
Unfortunately any modifications to HAOS are not possible by the nature its built (you can not install our kernel modules in a good manner, possible in theory, but very high risk that it will break by future HAOS update). But it should be possible to use our Bookworm image, install Evok and then install [Home Assistant supervised](https://github.com/home-assistant/supervised-installer). However, we can not guarantee it will work, this way is not supported by us.

closed 2024-06-20 by kratochvil01