UnipiTechnology/evok #116: unable to install beta version for test purposes

rudyvan, 2024-03-18:
trying to install the beta version on a raspberry 5 8GB with unipi 1.1
tried with lite and full versions, 32bit and 64bit, always same result that evok package is not found with apt-get

Has that to do with 2 times message 'Missing unipi-id module or bad id eprom' when installing with wget?

See below the installation history.

As soon as the installation works, i will try the websocket setup.
Thanks for your help.

```
pi@PI-Main:~ $ sudo su
root@PI-Main:/home/pi# wget -qO - https://repo.unipi.technology/debian/raspberry-unipi1.sh | bash
Running Raspberry Pi OS Debian/bookworm on arm64
Downloading and importing the Unipi repository signing key

Downloading the APT definition for Unipi repository
deb https://repo.unipi.technology/debian bookworm unipi1-main main

Updating the package list
Hit:1 http://deb.debian.org/debian bookworm InRelease
Hit:2 http://deb.debian.org/debian-security bookworm-security InRelease                                                
Hit:3 http://deb.debian.org/debian bookworm-updates InRelease                                                          
Hit:4 http://archive.raspberrypi.com/debian bookworm InRelease                                                         
Get:5 https://repo.unipi.technology/debian bookworm InRelease [124 kB]        
Get:6 https://repo.unipi.technology/debian bookworm/unipi1-main all Packages [1,935 B]
Get:7 https://repo.unipi.technology/debian bookworm/unipi1-main arm64 Packages [11.8 kB]
Get:8 https://repo.unipi.technology/debian bookworm/unipi1-main armhf Packages [8,975 B]
Get:9 https://repo.unipi.technology/debian bookworm/main armhf Packages [7,731 B]
Get:10 https://repo.unipi.technology/debian bookworm/main arm64 Packages [7,919 B]
Get:11 https://repo.unipi.technology/debian bookworm/main all Packages [3,996 B]
Fetched 167 kB in 1s (222 kB/s)
Reading package lists... Done
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following additional packages will be installed:
  libiio0 unipi-os-configurator
The following NEW packages will be installed:
  libiio0 unipi-kernel-modules unipi-one-modbus unipi-os-configurator unipi-os-configurator-data
0 upgraded, 5 newly installed, 0 to remove and 0 not upgraded.
Need to get 850 kB of archives.
After this operation, 3,845 kB of additional disk space will be used.
Get:1 http://deb.debian.org/debian bookworm/main arm64 libiio0 arm64 0.24-4 [47.1 kB]
Get:2 https://repo.unipi.technology/debian bookworm/main arm64 unipi-os-configurator arm64 0.54~bookworm [14.9 kB]
Get:3 https://repo.unipi.technology/debian bookworm/unipi1-main arm64 unipi-os-configurator-data arm64 0.42~bookworm-unipi1x64 [10.5 kB]
Get:4 https://repo.unipi.technology/debian bookworm/unipi1-main arm64 unipi-kernel-modules arm64 2.50.6.6.20-1+rpt1 [137 kB]
Get:5 https://repo.unipi.technology/debian bookworm/unipi1-main arm64 unipi-one-modbus arm64 0.20.3-pre~bookworm-unipi1x64 [641 kB]
Fetched 850 kB in 1s (1,235 kB/s)         
Selecting previously unselected package libiio0:arm64.
(Reading database ... 127435 files and directories currently installed.)
Preparing to unpack .../libiio0_0.24-4_arm64.deb ...
Unpacking libiio0:arm64 (0.24-4) ...
Selecting previously unselected package unipi-os-configurator.
Preparing to unpack .../unipi-os-configurator_0.54~bookworm_arm64.deb ...
Unpacking unipi-os-configurator (0.54~bookworm) ...
Selecting previously unselected package unipi-os-configurator-data.
Preparing to unpack .../unipi-os-configurator-data_0.42~bookworm-unipi1x64_arm64.deb ...
Adding 'diversion of /boot/overlays/ds2482.dtbo to /usr/share/rpikernelhack/overlays/ds2482.dtbo by rpikernelhack'
Adding 'diversion of /boot/overlays/em0018_slot1.dtbo to /usr/share/rpikernelhack/overlays/em0018_slot1.dtbo by rpikernelhack'
Adding 'diversion of /boot/overlays/em0018_slot2.dtbo to /usr/share/rpikernelhack/overlays/em0018_slot2.dtbo by rpikernelhack'
Adding 'diversion of /boot/overlays/em0018_slot3.dtbo to /usr/share/rpikernelhack/overlays/em0018_slot3.dtbo by rpikernelhack'
Adding 'diversion of /boot/overlays/em0018_slot4.dtbo to /usr/share/rpikernelhack/overlays/em0018_slot4.dtbo by rpikernelhack'
Adding 'diversion of /boot/overlays/em0018_slot5.dtbo to /usr/share/rpikernelhack/overlays/em0018_slot5.dtbo by rpikernelhack'
Adding 'diversion of /boot/overlays/em0018_slot6.dtbo to /usr/share/rpikernelhack/overlays/em0018_slot6.dtbo by rpikernelhack'
Adding 'diversion of /boot/overlays/em0018_slot7.dtbo to /usr/share/rpikernelhack/overlays/em0018_slot7.dtbo by rpikernelhack'
Adding 'diversion of /boot/overlays/unipi_id.dtbo to /usr/share/rpikernelhack/overlays/unipi_id.dtbo by rpikernelhack'
Adding 'diversion of /boot/overlays/unipi_rtc.dtbo to /usr/share/rpikernelhack/overlays/unipi_rtc.dtbo by rpikernelhack'
Adding 'diversion of /boot/overlays/unipi_unipi10.dtbo to /usr/share/rpikernelhack/overlays/unipi_unipi10.dtbo by rpikernelhack'
Adding 'diversion of /boot/overlays/unipi_unipi11.dtbo to /usr/share/rpikernelhack/overlays/unipi_unipi11.dtbo by rpikernelhack'
[... 92 lines cut ...]
The system will reboot now!

Broadcast message from root@PI-Main on pts/2 (Mon 2024-03-18 17:04:32 CET):

The system will reboot now!

root@PI-Main:/home/pi# Linux PI-Main 6.6.20+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.20-1+rpt1 (2024-03-07) aarch64

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Mon Mar 18 17:05:26 2024
pi@PI-Main:~ $ sudo apt-get install evok
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Package evok is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'evok' has no installation candidate
```

cewbdex, 2024-03-21:
Hello,
it looks like your kernel modules were installed correctly, your installation is now missing the correct evok repo. follow my steps:

`sudo nano /etc/apt/sources.list.d/unipi-evok.list`

place this in the file and then ctrl+o, enter and ctrl+x

`deb https://repo.unipi.technology/debian betaevok:bookworm main unipi1-main unipi1-test test`

Now run those commands:

`sudo apt update`

`sudo apt install evok evok-web`

Web is now installed separately, if you wish to use this feature.

rudyvan, 2024-03-21:
yep, installation works now.

but running into another issue, seems that the uni-pi is not detected and therefore autogen does not work:

```
(lucy) pi@PI-Main:~/lucy $ sudo /opt/unipi/tools/os-configurator -f
Missing unipi-id module or bad id eprom.

run-parts: executing /opt/unipi/os-configurator/run.d/10-config_txt.sh
run-parts: executing /opt/unipi/os-configurator/run.d/20-overlays.sh
run-parts: executing /opt/unipi/os-configurator/run.d/20-uoverlays.sh
run-parts: executing /opt/unipi/os-configurator/run.d/30-udev.sh
run-parts: executing /opt/unipi/os-configurator/run.d/30-udevvar.sh
run-parts: executing /opt/unipi/os-configurator/run.d/50-configure-one.sh
run-parts: executing /opt/unipi/os-configurator/run.d/50-unipi-hostname.sh
run-parts: executing /opt/unipi/os-configurator/run.d/98-bootcmd.sh
run-parts: executing /opt/unipi/os-configurator/run.d/99-wait-for-resize.sh
run-parts: executing /opt/unipi/os-configurator/run.d/evok-autogen.sh
Device not recognized!  ('UNIPI_PRODUCT_ID')
(lucy) pi@PI-Main:~/lucy $ ls /etc/evok
config.yaml  hw_definitions
```

I will try with a more regular setup as the current setup with a pimoroni with NVMe Base for Raspberry Pi 5 and unipi with os  64 bit might give issues.

kratochvil01, 2024-03-22:
Hello Rudyvan,

I have set up the same configuration of devices as you have with fresh Raspberry Pi OS Lite, and the installation of Evok and autogen works fine for me.

Your issue may lie in the fact that the script for modifying the boot configuration did not run properly. Please check if you have the line `include config_unipi.inc` in your boot configuration. You can view the file using the command `less /boot/firmware/config.txt`.

If the configuration is correct, then it's possible that the EEPROM on your device is damaged. Please check if the system correctly detected the EEPROM using the command `i2cdetect -y 1`. You may need to install the I2C tools using the command `apt install -y i2c-tools`. On Unipi 1 the EEPROM is located at address 0x50.
The output of this command should look like this:
[image]

rudyvan, 2024-03-22:
with i2cdetect i saw no devices appearing.
i supplied power directly (jumper 1 was unset) to unipi and then the detection worked.
Now the raspberry RPI5 and unipi are powered separately (by the same din rail power supply) and..
everything works!
Thanks!

closed 2024-03-22 by rudyvan