UnipiTechnology/evok #117: Beta release: inputs not working in webinterface and websocket

rudyvan, 2024-03-22:
Hello,

wanted to provide feedback on my tests with the beta release on a unipi1.1.
i modified config.yaml for port 8181 and nginx/sites-available/evok with api 8181 and server 8180

the web-interface on port 8180 responds and shows all inputs, relays, ..
my application with websocket on port 8181 works and gets all the feedback on cmd all (see below), also when items such as outputs or temperature sensors change.

what works:
- relays 1..8 : works through web-interface and through api websocket
- 1w devices : idem

what doesn't seem to work:
- ao:  works through web-interface and api, but with a volt meter to output stays zero (i applied ground to pin AOG and 12v to pin AOV)
- inputs:  they show in the web-interface but they don't change value when physically triggered (with a wire to the 12V pin)

not tested yet:
- ai1 and ai2
- other extension boards added

Any reason why input changes are not detected?
strangely in the web-interface, all inputs show ON, only input 1_03 shows OFF 

```
2024-03-22 14:02:25 PI-Main:/home/pi/lucy/lucy_app/drivers/hardware.py@1026:!!unipi Decode, ?device? [{'dev': 'relay', 'circuit': '1_01', 'value': 0, 'glob_dev_id': 2}, {'dev': 'relay', 'circuit': '1_02', 'value': 0, 'glob_dev_id': 2}, {'dev': 'relay', 'circuit': '1_03', 'value': 0, 'glob_dev_id': 2}, 
{'dev': 'relay', 'circuit': '1_04', 'value': 0, 'glob_dev_id': 2}, {'dev': 'relay', 'circuit': '1_05', 'value': 0, 'glob_dev_id': 2}, {'dev': 'relay', 'circuit': '1_06', 'value': 0, 'glob_dev_id': 2}, {'dev': 'relay', 'circuit': '1_07', 'value': 0, 'glob_dev_id': 2}, {'dev': 'relay', 'circuit': '1_08', 
'value': 0, 'glob_dev_id': 2}, {'dev': 'input', 'circuit': '1_01', 'value': 1, 'debounce': 0, 'counter_modes': ['Enabled', 'Disabled'], 'counter_mode': 'Enabled', 'counter': 0, 'mode': 'Simple', 'modes': ['Simple'], 'glob_dev_id': 2}, {'dev': 'input', 'circuit': '1_02', 'value': 1, 'debounce': 0, 
'counter_modes': ['Enabled', 'Disabled'], 'counter_mode': 'Enabled', 'counter': 0, 'mode': 'Simple', 'modes': ['Simple'], 'glob_dev_id': 2}, {'dev': 'input', 'circuit': '1_03', 'value': 0, 'debounce': 0, 'counter_modes': ['Enabled', 'Disabled'], 'counter_mode': 'Enabled', 'counter': 135, 'mode': 
'Simple', 'modes': ['Simple'], 'glob_dev_id': 2}, {'dev': 'input', 'circuit': '1_04', 'value': 1, 'debounce': 0, 'counter_modes': ['Enabled', 'Disabled'], 'counter_mode': 'Enabled', 'counter': 0, 'mode': 'Simple', 'modes': ['Simple'], 'glob_dev_id': 2}, {'dev': 'input', 'circuit': '1_05', 'value': 1, 
'debounce': 0, 'counter_modes': ['Enabled', 'Disabled'], 'counter_mode': 'Enabled', 'counter': 0, 'mode': 'Simple', 'modes': ['Simple'], 'glob_dev_id': 2}, {'dev': 'input', 'circuit': '1_06', 'value': 1, 'debounce': 0, 'counter_modes': ['Enabled', 'Disabled'], 'counter_mode': 'Enabled', 'counter': 69, 
'mode': 'Simple', 'modes': ['Simple'], 'glob_dev_id': 2}, {'dev': 'input', 'circuit': '1_07', 'value': 1, 'debounce': 0, 'counter_modes': ['Enabled', 'Disabled'], 'counter_mode': 'Enabled', 'counter': 0, 'mode': 'Simple', 'modes': ['Simple'], 'glob_dev_id': 2}, {'dev': 'input', 'circuit': '1_08', 
'value': 1, 'debounce': 0, 'counter_modes': ['Enabled', 'Disabled'], 'counter_mode': 'Enabled', 'counter': 0, 'mode': 'Simple', 'modes': ['Simple'], 'glob_dev_id': 2}, {'dev': 'input', 'circuit': '1_09', 'value': 1, 'debounce': 0, 'counter_modes': ['Enabled', 'Disabled'], 'counter_mode': 'Enabled', 
'counter': 0, 'mode': 'Simple', 'modes': ['Simple'], 'glob_dev_id': 2}, {'dev': 'input', 'circuit': '1_10', 'value': 1, 'debounce': 0, 'counter_modes': ['Enabled', 'Disabled'], 'counter_mode': 'Enabled', 'counter': 0, 'mode': 'Simple', 'modes': ['Simple'], 'glob_dev_id': 2}, {'dev': 'input', 'circuit': 
'1_11', 'value': 1, 'debounce': 0, 'counter_modes': ['Enabled', 'Disabled'], 'counter_mode': 'Enabled', 'counter': 0, 'mode': 'Simple', 'modes': ['Simple'], 'glob_dev_id': 2}, {'dev': 'input', 'circuit': '1_12', 'value': 1, 'debounce': 0, 'counter_modes': ['Enabled', 'Disabled'], 'counter_mode': 
'Enabled', 'counter': 0, 'mode': 'Simple', 'modes': ['Simple'], 'glob_dev_id': 2}, {'dev': 'ai', 'circuit': '1_01', 'value': 0.001, 'unit': 'V', 'glob_dev_id': 2, 'mode': 'Voltage', 'modes': {'Voltage': {'unit': 'V', 'range': [0, 10]}}, 'range': [0, 10]}, {'dev': 'ai', 'circuit': '1_02', 'value': 0.002,
'unit': 'V', 'glob_dev_id': 2, 'mode': 'Voltage', 'modes': {'Voltage': {'unit': 'V', 'range': [0, 10]}}, 'range': [0, 10]}, {'dev': 'ao', 'circuit': '1_01', 'mode': 'Voltage', 'modes': {'Voltage': {'unit': 'V', 'range': [0, 10]}}, 'glob_dev_id': 2, 'value': 10.0, 'unit': 'V', 'range': [0, 10]}, {'dev': 
'temp', 'circuit': '28CC0560070000C4', 'address': '28.CC0560070000.C4', 'value': 22.7, 'lost': True, 'time': 749.669302308, 'interval': 10, 'typ': 'DS18B20'}, {'dev': 'owbus', 'circuit': 'OWFS', 'bus': 'via OWFS', 'scan_interval': 60, 'interval': 10, 'do_scan': False, 'do_reset': False}, {'dev': 
'modbus_slave', 'circuit': '1', 'glob_dev_id': 2, 'last_comm': 0.0027756690979003906, 'slave_id': 0, 'modbus_type': 'TCP', 'scan_interval': 0.02}]
```

rudyvan, 2024-03-22:
further, i tested analog input 1 and 2 and they work.
digital inputs and analog output not

miroslav-ondra, 2024-03-25:
I assume you have Raspberry Pi 5. On RPi 5 are pins on externel header  connected to other gpio chip. This m

miroslav-ondra, 2024-03-26:
There is new version of package unipi-one-modbus and unipio-os-configurator-data, which solves changing configs for RPI3/4/5
Use `apt update && apt upgrade`
DI and AO on RPi5 should be working after autoreconfiguration