UnipiTechnology/evok #118: Beta release: how to add extension boards via i2c

rudyvan, 2024-03-22:
Hello,

i have added an EMO R8 relais extension board on hex 21 and a EMO-1w 8 extension board on hex 19, see below the i2cdetect output that shows they are recognised.

[image]

I assumed they were automatically recognised, i tried a reinstallation of evok and a reboot but unfortunately to no avail.

Should i therefore update a yaml file in /etc/evok? 
The examples in the docs are for modbus or tcp extensions, but not for i2c legacy devices?
How can we add i2c legacy devices?  Please point me in the right direction.

miroslav-ondra, 2024-03-26:
Look at [https://github.com/UniPiTechnology/unipi-one-modbus](url) how to add EMO R8.
EMO-1w is no more supported but Evok uses owserver. You can add second ow-master chip at 0x19 to config file of owserver.

rudyvan, 2024-03-26:
Hello,

To add an EMO R8 on address x21, i have followed the steps outlined.
see below, i added overlay in boot.txt, created module_id.1 and run os-configurator and HAS_EMOR8_1='1' emerged.

after reboot, i do not see the extra relays appearing in web-evok, is there something else to do to make them appear?

Thanks!

```
pi@PI-Main:~ $ sudo su
root@PI-Main:/home/pi# echo 0018 > /etc/unipi-id/module_id.1
root@PI-Main:/home/pi# exit
exit
pi@PI-Main:~ $ ls /etc/unipi-id
module_id.1  README.md
pi@PI-Main:~ $ sudo nano /etc/unipi-id/README.md
pi@PI-Main:~ $ sudo /opt/unipi/tools/os-configurator --force
DT='unipi_unipi11 em0018_slot1'
UDEV='unipi_unipi11 em0018_slot1'
HAS_DS2482='1'
HAS_RTC='1'
USE_ETC_MODULES='1'
UNIPI_PLATFORM='0101'
UNIPI_PRODUCT_ID='0101'
UNIPI_PRODUCT_NAME='UNIPI11'
UNIPI_PRODUCT_SERIAL='391708672'
HAS_EMOR8_1='1'
run-parts: executing /opt/unipi/os-configurator/run.d/10-config_txt.sh
run-parts: executing /opt/unipi/os-configurator/run.d/20-overlays.sh
run-parts: executing /opt/unipi/os-configurator/run.d/20-uoverlays.sh
run-parts: executing /opt/unipi/os-configurator/run.d/30-udev.sh
run-parts: executing /opt/unipi/os-configurator/run.d/30-udevvar.sh
run-parts: executing /opt/unipi/os-configurator/run.d/50-configure-one.sh
run-parts: executing /opt/unipi/os-configurator/run.d/50-unipi-hostname.sh
run-parts: executing /opt/unipi/os-configurator/run.d/98-bootcmd.sh
run-parts: executing /opt/unipi/os-configurator/run.d/99-wait-for-resize.sh
run-parts: executing /opt/unipi/os-configurator/run.d/evok-autogen.sh
Detect device UNIPI1 UNIPI11 (id=0x101, sn=391708672) with boards ['UNIPI11']
Reboot system to apply all changes in configuration
pi@PI-Main:~ $ sudo reboot
```

rudyvan, 2024-03-26:
Hello,

to have the legacy EMO-1w working, one can make the following changes to /etc/owfs.conf:

```
### CONFIGURED BY EVOK ###
server: i2c=/dev/i2c-1
server: w1
##########################
```

change to:

```
### CONFIGURED BY EVOK ###
server: i2c=ALL:ALL
server: w1
##########################
```

and then:

sudo service owserver restart

(just make sure the hex address of added EMO-1w start at x19)

rudyvan, 2024-03-26:
there seems to be an issue in os-configurator when adding an EMO-R8.

when i run os-configurator, it says that HAS_EMOR8_1='1'

```
DT='unipi_unipi11 em0018_slot1'
UDEV='unipi_unipi11 em0018_slot1'
HAS_DS2482='1'
HAS_RTC='1'
USE_ETC_MODULES='1'
UNIPI_PLATFORM='0101'
UNIPI_PRODUCT_ID='0101'
UNIPI_PRODUCT_NAME='UNIPI11'
UNIPI_PRODUCT_SERIAL='391708672'
HAS_EMOR8_1='1'
run-parts: executing /opt/unipi/os-configurator/run.d/10-config_txt.sh
run-parts: executing /opt/unipi/os-configurator/run.d/20-overlays.sh
run-parts: executing /opt/unipi/os-configurator/run.d/20-uoverlays.sh
run-parts: executing /opt/unipi/os-configurator/run.d/30-udev.sh
run-parts: executing /opt/unipi/os-configurator/run.d/30-udevvar.sh
run-parts: executing /opt/unipi/os-configurator/run.d/50-configure-one.sh
run-parts: executing /opt/unipi/os-configurator/run.d/50-unipi-hostname.sh
run-parts: executing /opt/unipi/os-configurator/run.d/98-bootcmd.sh
run-parts: executing /opt/unipi/os-configurator/run.d/99-wait-for-resize.sh
run-parts: executing /opt/unipi/os-configurator/run.d/evok-autogen.sh
Detect device UNIPI1 UNIPI11 (id=0x101, sn=391708672) with boards ['UNIPI11']
```

however /etc/autogen.yaml has nothing of EMO-R8:

```
omm_channels:
  LOCAL_TCP:
    type: MODBUSTCP
    hostname: 127.0.0.1
    port: 503
    device_info:
      family: UNIPI1
      model: UNIPI11
      sn: 391708672
      board_count: 1
    devices:
      1:
        slave-id: 0
        model: 'UNIPI11'
  OWFS:
    type: OWBUS
    interval: 10
    scan_interval: 60
```

if i manually add the information that i thing should be generated into /etc/autogen.yaml:

```
comm_channels:
  LOCAL_TCP:
    type: MODBUSTCP
    hostname: 127.0.0.1
    port: 503
    device_info:
      family: UNIPI1
      model: UNIPI11
      sn: 391708672
      board_count: 1
    devices:
      1:
        slave-id: 0
        model: 'UNIPI11'
      2:
        slave-id: 0
        model: 'EMO-R8'
  OWFS:
    type: OWBUS
    interval: 10
    scan_interval: 60
```

and i restart evok, then i see the extra relays in evok-web, but they still do not work
any idea, suggestions?

miroslav-ondra, 2024-03-26:
Change config:
      2:
        slave-id: **1**
        model: 'EMO-R8'

Emo-R8 is sitting on modbus address 1

miroslav-ondra, 2024-03-26:
Autoconfiguration of EMO-R8 for Evok is still missing :-(

rudyvan, 2024-03-26:
i updated autogen.yaml as per your instruction.
(did not know what slave-id is for, was not in the documentation)

```
comm_channels:
  LOCAL_TCP:
    type: MODBUSTCP
    hostname: 127.0.0.1
    port: 503
    device_info:
      family: UNIPI1
      model: UNIPI11
      sn: 391708672
      board_count: 2
    devices:
      1:
        slave-id: 0
        model: 'UNIPI11'
      2:
        slave-id: 1
        model: 'EMO-R8'
  OWFS:
    type: OWBUS
    interval: 10
    scan_interval: 60
```

shows ok in evok-web, but changing the relays does not work on the board for some reason.
Might it be that i am on a RPI5?   I did the app update and upgrade first, so i am with the latest unipi-one-modbus version.

miroslav-ondra, 2024-03-26:
What reports i2cdetect on 0x21 ?
Look to dmesg or  journalctl after changing relay status. Is there something strange?

rudyvan, 2024-03-27:
Indeed, there is an issue with the driver registration (via dmesg):

```
[    1.174196] i2c i2c-1: Failed to register i2c client mcp23008 at 0x21 (-16)
[    1.181357] i2c i2c-1: of_i2c: Failure registering /axi/pcie@120000/rp1/i2c@74000/mcp23008@21
[    1.189968] i2c i2c-1: Failed to create I2C device for /axi/pcie@120000/rp1/i2c@74000/mcp23008@21
[    1.203314] mcp230xx 1-0021: prop pinctrl-0 index 0 invalid phandle

```

although i2detect confirms the board is there:

```
(lucy) pi@PI-Main:~/lucy $ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- 18 19 -- -- -- -- -- -- 
20: UU 21 -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: UU -- -- -- -- -- -- 57 -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- UU -- -- -- -- -- -- UU 
70: -- -- -- -- -- -- -- --
```

what could the issue be?
should i try with an RPI4 instead of RPI5?

rudyvan, 2024-03-27:
tried with an RPI4 and same result:

```
[    2.301053] i2c i2c-1: Failed to register i2c client mcp23008 at 0x21 (-16)
[    2.301179] i2c i2c-1: of_i2c: Failure registering /soc/i2c@7e804000/mcp23008@21
[    2.301269] i2c i2c-1: Failed to create I2C device for /soc/i2c@7e804000/mcp23008@21
```

stuck at this point.

miroslav-ondra, 2024-03-27:
Remove please row from config.txt like 
dtoverlay=mcp23017,mcp23008,addr=0x21,noints
Overlay distributed by RaspiOS is  not corrently in case of noints
There is em0018_slotX overlay in config_unipi.txt

miroslav-ondra, 2024-03-27:
Also remove from config.txt rows like 
`dtoverlay=pwm,pin=18,2`
`dtoverlay=mcp342x,mcp3422,addr=0x68`
All this things are covered by unipi overlays if you have **unipi-os-configurator**

closed 2024-03-27 by rudyvan