UnipiTechnology/evok #120: I can't install Evok on fresh install of Raspbian

BucuresteanuPC, 2024-03-26:
### Prerequisites

* [Yes ] Are you running the latest Master or the latest release version?
* [Yes ] Did you try to reinstall EVOK?
* [Yes ] Did you perform a cursory search on the [forum] and [google]?

### Log files needed

Include the output of the following commands on your Neuron/UniPi device:

```
sudo su
cat /var/log/evok.access.log
ps -fax
cat /var/log/evok.log
```
I don't have this files on the rpi.

### Description
[Description of the bug or feature]
I try to install evok on fresh install Raspbian Stretch / Stretch Lite / Buster / Buster Lite / Bullseye... several versions of each with no succes, because i think the Evok isn't compatible with the new versions, but with no succes. I must to mention that we have a Raspberry Pi 3 B+ Version with Unipi 1.1 and Unipi 1.1 Lite versions.

The message i got was:

  Installing build dependencies ... done
    Complete output from command python setup.py egg_info:
    running egg_info
    creating pip-egg-info/PyYAML.egg-info
    writing pip-egg-info/PyYAML.egg-info/PKG-INFO
    writing top-level names to pip-egg-info/PyYAML.egg-info/top_level.txt
    writing dependency_links to pip-egg-info/PyYAML.egg-info/dependency_links.txt
    writing manifest file 'pip-egg-info/PyYAML.egg-info/SOURCES.txt'
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-install-OAS5c5/pyyaml/setup.py", line 295, in <module>
        python_requires='>=2.7, !=3.0.*, !=3.1.*, !=3.2.*, !=3.3.*, !=3.4.*, !=3.5.*',
      File "/tmp/pip-build-env-fLDcsE/lib/python2.7/site-packages/setuptools/__init__.py", line 162, in setup
        return distutils.core.setup(**attrs)
      File "/usr/lib/python2.7/distutils/core.py", line 151, in setup
        dist.run_commands()
      File "/usr/lib/python2.7/distutils/dist.py", line 953, in run_commands
        self.run_command(cmd)
      File "/usr/lib/python2.7/distutils/dist.py", line 972, in run_command
        cmd_obj.run()
      File "/tmp/pip-build-env-fLDcsE/lib/python2.7/site-packages/setuptools/command/egg_info.py", line 296, in run
        self.find_sources()
      File "/tmp/pip-build-env-fLDcsE/lib/python2.7/site-packages/setuptools/command/egg_info.py", line 303, in find_sources
        mm.run()
      File "/tmp/pip-build-env-fLDcsE/lib/python2.7/site-packages/setuptools/command/egg_info.py", line 534, in run
        self.add_defaults()
      File "/tmp/pip-build-env-fLDcsE/lib/python2.7/site-packages/setuptools/command/egg_info.py", line 570, in add_defaults
        sdist.add_defaults(self)
      File "/tmp/pip-build-env-fLDcsE/lib/python2.7/site-packages/setuptools/command/py36compat.py", line 36, in add_defaults
        self._add_defaults_ext()
      File "/tmp/pip-build-env-fLDcsE/lib/python2.7/site-packages/setuptools/command/py36compat.py", line 120, in _add_defaults_ext
        self.filelist.extend(build_ext.get_source_files())
      File "/tmp/pip-install-OAS5c5/pyyaml/setup.py", line 201, in get_source_files
        self.cython_sources(ext.sources, ext)
      File "/usr/lib/python2.7/distutils/cmd.py", line 105, in __getattr__
        raise AttributeError, attr
    AttributeError: cython_sources

    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-install-OAS5c5/pyyaml/

And when I try to run the test file:
"
from jsonrpclib import Server
s=Server("http://localhost:8181/rpc")
s.relay_set(1,1)
s.relay_get(1)
s.relay_set(1,0)
s.relay_get(0)
s.ai_get(1)
"

root@Matrita4gb:/home/pi/Desktop# python test.py
Traceback (most recent call last):
  File "test.py", line 1, in <module>
    from jsonrpclib import Server
ImportError: No module named jsonrpclib

root@Matrita4gb:/home/pi/Desktop# python3 test.py
Traceback (most recent call last):
  File "test.py", line 1, in <module>
    from jsonrpclib import Server
ModuleNotFoundError: No module named 'jsonrpclib'

I try to update/upgrade pip and pip3, i try to chage the default python to python 2, i try to make different tricks with no succes.
We have more Unipi 1.1 and Lite versions on more sites and we stuck with old os with more packages that what we need, we have backups from it and to new site we have to use the old images because we stuck on it.
Sometimes the jsonrpclib is installed, but evok don't work, and the input/output don't show on the web interface.
What I make wrong?

### Steps to Reproduce
sudo su
apt update --allow-releaseinfo-change
apt-get update --allow-releaseinfo-change
apt upgrade
apt-get upgrade
reboot

sudo su
wget https://github.com/UniPiTechnology/evok/archive/2.4.34.zip
unzip 2.4.34.zip
chmod 777 evok-2.4.34
cd evok-2.4.34
chmod 777 *
bash install-evok.sh

kratochvil01, 2024-03-26:
Hello, It seems that the problem may be caused by the loss of support for some of the required dependencies. I will try to solve this issue.

Nevertheless, a new version of Evok will be released shortly, which is currently in beta. If you are developing a new application or updating an existing one, we recommend considering transitioning to the new version of Evok. You can see more information [here](https://github.com/UniPiTechnology/evok/blob/dev-bookworm/README.md).

kratochvil01, 2024-06-20:
Evok v3.0.1 is released.

closed 2024-06-20 by kratochvil01