UnipiTechnology/evok #122: A request to change the debounce, counter, etc. returns the original value

vytun, 2024-03-26:
When sending a API request to set a debounce, counter, etc., the original value is returned in the API response (tested with WS, REST). This is probably a **feature**, since the value is already written via Modbus at the time of receiving the API response, but has not yet been read from the device (via Modbus).

Solution: mention this in the documentation in the "Good to know" section.

kratochvil01, 2024-03-26:
Evok returns the last read values. The request to set the value is a blocking operation that takes place during modbus writing, but after the writing is finished, reading is not performed immediately. So the status of the device, which Evok returns immediately after writing, is not yet updated.

Evok regularly reads all devices (based on [configuration](https://evok.readthedocs.io/en/latest/configs/evok_configuration/#hardware-configuration)). The updated status can therefore be obtained even after the Evok has read. To check the status change, We recommend using a [websocket](https://evok.readthedocs.io/en/latest/apis/websocket/) that sends an event in case of a status change.

matoushybl, 2026-10-08:
Duplicate of #74

closed 2026-10-08 by matoushybl