UnipiTechnology/evok #129: Beta Release:  Digital Inputs not working on RPi4 and Unipi Lite

rudyvan, 2024-03-28:
just done a fresh install, and surprise for something working before doesn't work any more.
Installing evok on RPi4 with Unipi lite.

/boot/firmware/config.txt:
```
include config_unipi.inc
...
```

/boot/firmware/config_unipi.inc:
```
include config_unipi.inc
# For more options and information see
# http://rptl.io/configtxt
# Some settings may impact device functionality. See link above for details

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
```

/etc/evok/autogen.yaml
```
comm_channels:
  LOCAL_TCP:
    type: MODBUSTCP
    hostname: 127.0.0.1
    port: 503
    device_info:
      family: UNIPI1
      model: UNIPI1L
      sn: 1359020032
      board_count: 1
    devices:
      1:
        slave-id: 0
        model: 'UNIPI11LITE'
  OWFS:
    type: OWBUS
    interval: 10
    scan_interval: 60
```

Maybe the DI mappings are for an RPi5 and not for RPi4?
dmesg does not show errors

kratochvil01, 2024-03-28:
Hello,

Do you have 'unipi-one-modbus' configured correctly? For raspberry pi 5 must be used 'gpiochip4' (as mentioned [here](https://github.com/UniPiTechnology/evok/issues/117#issuecomment-2019122127)), but for recent raspberry pi versions must be used 'gpiochip0'.

rudyvan, 2024-03-28:
well, i have just done a fresh install without considering or changing unipi-one-modbus as i assumed installing evok and evok-web with apt (after wget) takes care of everything, auto configuring unipi lite, etc..

and indeed, i assume a wrongly file appeared in:   /etc/unipi-one-modbus.d/5-unipi-lite.yaml

why this appears there while installing on an raspberry 4?
are you assuming that at every installation, the customer has to check /etc/unipi-one-modbus.d ?

I also do not find in what yaml file i have to modify the  chip_path: /dev/gpiochip0

```
ls /etc/unipi-one-modbus.d/

1-modbus.yaml  
5-unipi-lite.yaml
```

It is difficult for a novice to go about without training or updated doc.
Please indicate where to update the chip_path?

rudyvan, 2024-04-01:
I have done some further investigation and looked into the file:

/usr/share/unipi-one-modbus/5-unipi-lite.yaml

```yaml
-   type: modbus.slavemap
    channel: *main_server
    slaveid: 0
    childs:
    -   type: modbus.registers
        mode: registers
        devs: [ REx6, DI1, DI2, DI3, DI4, DI5, DI6 ]
    -   type: modbus.registers
        mode: coils
        devs: [ REx6, DI1, DI2, DI3, DI4, DI5, DI6 ]

-   name: GPIODI
    type: gpiod.gpionotify
    chip_path: /run/unipi-plc/by-sys/DI/gpiochip
    childs:
        - { name: DI1,  type: gpiod.input, pin: 4,  input: 8,  register: 0, reg_bit: 0,  debounce_reg: 1010, counter_reg: 2, debounce: 0 }
        - { name: DI2,  type: gpiod.input, pin: 17, input: 9,  register: 0, reg_bit: 1,  debounce_reg: 1011, counter_reg: 4, debounce: 0  }
        - { name: DI3,  type: gpiod.input, pin: 27, input: 10, register: 0, reg_bit: 2,  debounce_reg: 1012, counter_reg: 6, debounce: 0 }
        - { name: DI4,  type: gpiod.input, pin: 23, input: 11, register: 0, reg_bit: 3,  debounce_reg: 1013, counter_reg: 8, debounce: 0 }
        - { name: DI5,  type: gpiod.input, pin: 22, input: 12, register: 0, reg_bit: 4,  debounce_reg: 1014, counter_reg: 10, debounce: 0 }
        - { name: DI6,  type: gpiod.input, pin: 24, input: 13, register: 0, reg_bit: 5,  debounce_reg: 1015, counter_reg: 12, debounce: 0 }

-   name: GPIORO
    type: gpiod.gpiochip
    chip_path: /run/unipi-plc/by-sys/RO/gpiochip
    childs:
        -   name: REx6
            type: gpiod.regcoils
            start_coil: 0
            register: 1
            pins: [ 7, 6, 5, 4, 3, 2 ]

```

**however the device file for the digital inputs is missing:
/run/unipi-plc/by-sys/DI/gpiochip**

The device file for the relays is present.
/run/unipi-plc/by-sys/RO/gpiochip

Why would the device file for DI not be installed?

Thanks for your help!

miroslav-ondra, 2024-04-02:
Problem is unipi-os-configurator-data. Copy&paste error in udev rules for unipi-lite on RPi4. It will be fixed soon in next release of package

rudyvan, 2024-04-02:
ok, you mean the following file is wrong?

/opt/unipi/os-configurator/udev/unipi_unipi1l.rules

```
ACTION=="add|remove", SUBSYSTEM=="gpio", DEVPATH=="*/soc/3f200000.gpio/gpiochip*",  RUN+="/opt/unipi/tools/uhelper DI gpiochip 0 $name"
ACTION=="add|remove", SUBSYSTEM=="gpio", DEVPATH=="*/soc/3f200000.gpio/gpiochip*",  RUN+="/opt/unipi/tools/uhelper DI gpiochip 0 $name"
ACTION=="add|remove", SUBSYSTEM=="gpio", DEVPATH=="*/axi/1000120000.pcie/1f000d0000.gpio/gpiochip*",  RUN+="/opt/unipi/tools/uhelper DI gpiochip 0 $name"
ACTION=="add|remove", SUBSYSTEM=="gpio", DEVPATH=="*/i2c-1/1-0020/gpiochip*",  RUN+="/opt/unipi/tools/uhelper RO gpiochip 0 $name"
```

what do i need to change here or how can i update an existing installation to unipi-os-configurator-data_0.50?
If i change the file, do i simply have to reboot or run something else?   does running /opt/unipi/tools/os-configurator overwrite the change?

A little bit more patience here for me, i am not an expert in the new evok setup, but i have 5 unipi lite configurations to test with the beta evok release.     All 10 unipi1.1 configurations i have are updated and work as a charm.

miroslav-ondra, 2024-04-02:
The simplest way is apt update && apt install unipi-os-configurator-data
Os-configurator is started in postinst script automatically,  you need to reboot manually.

rudyvan, 2024-04-02:
works, thanks!

closed 2024-04-02 by cewbdex