UnipiTechnology/evok #130: Invalid device circuit number 1_01

GlennGeenen, 2024-04-01:
### Prerequisites

* [x] Are you running the latest Master or the latest release version?
* [x] Did you try to reinstall EVOK?
* [x] Did you perform a cursory search on the [forum] and [google]?

### Log files needed

```
[I 240401 08:22:38 config:78] eprom: UniPi Neuron L203 version: 1.4 serial: 0x59
[I 240401 08:22:38 evok:1899] Starting using config file /etc/evok.conf
[I 240401 08:22:38 config:110] YAML Definition loaded: /etc/hw_definitions/BuiltIn/L203.yaml, type: UniPiBuiltIn
[I 240401 08:22:39 config:103] YAML Definition loaded: xS40.yaml, type: 3, definition count 0
[I 240401 08:22:39 config:103] YAML Definition loaded: IC-Di1Ro2-1.yaml, type: 3, definition count 1
[I 240401 08:22:39 config:103] YAML Definition loaded: IC-Light1P-1.yaml, type: 3, definition count 2
[I 240401 08:22:39 config:103] YAML Definition loaded: xS11.yaml, type: 3, definition count 3
[I 240401 08:22:39 config:103] YAML Definition loaded: ID-AiUC8-1.yaml, type: 3, definition count 4
[I 240401 08:22:39 config:103] YAML Definition loaded: IAQ.yaml, type: 3, definition count 5
[I 240401 08:22:39 config:103] YAML Definition loaded: xS30.yaml, type: 3, definition count 6
[I 240401 08:22:39 config:103] YAML Definition loaded: xG18.yaml, type: 3, definition count 7
[I 240401 08:22:39 config:103] YAML Definition loaded: ID-Do16-1.yaml, type: 3, definition count 8
[I 240401 08:22:39 config:103] YAML Definition loaded: xS10.yaml, type: 3, definition count 9
[I 240401 08:22:39 config:103] YAML Definition loaded: IC-DiS2Do2-1.yaml, type: 3, definition count 10
[I 240401 08:22:39 config:103] YAML Definition loaded: xS50.yaml, type: 3, definition count 11
[I 240401 08:22:39 config:103] YAML Definition loaded: CUSTOM_MODBUS_DEVICE.yaml, type: 3, definition count 12
[I 240401 08:22:39 config:103] YAML Definition loaded: ID-DiS16-1.yaml, type: 3, definition count 13
[I 240401 08:22:39 config:103] YAML Definition loaded: IC-AiUC8-2.yaml, type: 3, definition count 14
[I 240401 08:22:39 config:103] YAML Definition loaded: xS51.yaml, type: 3, definition count 15
[I 240401 08:22:39 config:103] YAML Definition loaded: evok-alias.yaml, type: 2, definition count 0
[I 240401 08:22:39 evok:1969] HTTP server listening on port: 8080
[I 240401 08:22:39 modbusclient_tornado:134] TCP/Modbus client to 127.0.0.1:502 started
[I 240401 08:22:39 neuron:215] Reading SPI boards
[I 240401 08:22:39 neuron:230] No board on SPI 1
[I 240401 08:22:39 neuron:230] No board on SPI 2
[I 240401 08:22:39 neuron:230] No board on SPI 3
[E 240401 08:22:39 config:421] Invalid device circuit number 1_01
    Traceback (most recent call last):
      File "/opt/evok/lib/python2.7/site-packages/evok/config.py", line 416, in add_aliases
        dev_obj = Devices.by_int(dev_pointer["dev_type"], dev_pointer["circuit"])
      File "/opt/evok/lib/python2.7/site-packages/evok/devices.py", line 68, in by_int
        raise Exception('Invalid device circuit number %s' % str(circuit))
    Exception: Invalid device circuit number 1_01
[E 240401 08:22:39 config:421] Invalid device circuit number 1_02
    Traceback (most recent call last):
      File "/opt/evok/lib/python2.7/site-packages/evok/config.py", line 416, in add_aliases
        dev_obj = Devices.by_int(dev_pointer["dev_type"], dev_pointer["circuit"])
      File "/opt/evok/lib/python2.7/site-packages/evok/devices.py", line 68, in by_int
        raise Exception('Invalid device circuit number %s' % str(circuit))
    Exception: Invalid device circuit number 1_02
[I 240401 08:22:39 owclient:450] Entering OWW loop with PID 703

```

```
  PID TTY      STAT   TIME COMMAND
    2 ?        S      0:00 [kthreadd]
    3 ?        I<     0:00  \_ [rcu_gp]
    4 ?        I<     0:00  \_ [rcu_par_gp]
    6 ?        I<     0:00  \_ [kworker/0:0H-mmc_complete]
    8 ?        I<     0:00  \_ [mm_percpu_wq]
    9 ?        S      0:00  \_ [rcu_tasks_rude_]
   10 ?        S      0:00  \_ [rcu_tasks_trace]
   11 ?        S      0:00  \_ [ksoftirqd/0]
   12 ?        I      0:00  \_ [rcu_sched]
   13 ?        S      0:00  \_ [migration/0]
   14 ?        S      0:00  \_ [cpuhp/0]
   15 ?        S      0:00  \_ [cpuhp/1]
   16 ?        S      0:00  \_ [migration/1]
   17 ?        S      0:00  \_ [ksoftirqd/1]
   20 ?        S      0:00  \_ [cpuhp/2]
[... 108 lines cut ...]
  712 ?        Ss     0:02 /opt/evok/bin/python /opt/evok/lib/python2.7/site-packages/evok/evok.py
  713 ?        Sl     0:00  \_ /opt/evok/bin/python /opt/evok/lib/python2.7/site-packages/evok/evok.py

```

### Description

Inputs/Outputs dashboard empty. Error in the logs when I interact with NodeRed:
```
[E 240401 08:22:39 config:421] Invalid device circuit number 1_02
    Traceback (most recent call last):
      File "/opt/evok/lib/python2.7/site-packages/evok/config.py", line 416, in add_aliases
        dev_obj = Devices.by_int(dev_pointer["dev_type"], dev_pointer["circuit"])
      File "/opt/evok/lib/python2.7/site-packages/evok/devices.py", line 68, in by_int
        raise Exception('Invalid device circuit number %s' % str(circuit))
    Exception: Invalid device circuit number 1_02
```

### Steps to Reproduce

1. Installed latest Node-RED image: https://kb.unipi.technology/en:hw:02-neuron:download-image:02-nodered on a RPi4 on Neuron L203
2. Updated to latest evok version which makes no difference

[forum]:https://forum.unipi.technology/category/4/official-evok-api
[google]:http://www.google.com/

miroslav-ondra, 2024-04-02:
Can you send us output of dmesg? Important section about spi /unipi devices.

GlennGeenen, 2024-04-05:
i disabled the wifi so the brcmfmac errors disappeared. No red lines, only a bold message:
**Kernel parameter elevator= does not have any effect anymore.
Please use sysfs to set IO scheduler for individual devices.**

kratochvil01, 2024-04-05:
Hello
It looks like Hardware problem. The device cannot establish a connection with any section. Did this device work with another OS?

You can try [beta version](https://evok.readthedocs.io/en/latest/installation/#node-red-beta-os-images) with newer drivers.

GlennGeenen, 2024-04-05:
beta version shows errors on boot:
[    5.906682] unipi-spi spi0.3: spi channel looks empty
[    5.910061] unipi-spi spi0.2: spi channel looks empty
[    5.910789] unipi-spi spi0.1: spi channel looks empty

evok log:
INFO:evok:Starting Evok v3.0.1 using config directory '/etc/evok'.
INFO:evok:Setting logging level to 'DEBUG'.
DEBUG:evok:YAML Definition loaded: /etc/evok/hw_definitions/13.yaml, definition count 0
DEBUG:evok:YAML Definition loaded: /etc/evok/hw_definitions/00.yaml, definition count 1
DEBUG:evok:YAML Definition loaded: /etc/evok/hw_definitions/xS50.yaml, definition count 2
DEBUG:evok:YAML Definition loaded: /etc/evok/hw_definitions/xG18.yaml, definition count 3
DEBUG:evok:YAML Definition loaded: /etc/evok/hw_definitions/07.yaml, definition count 4
DEBUG:evok:YAML Definition loaded: /etc/evok/hw_definitions/08.yaml, definition count 5
DEBUG:evok:YAML Definition loaded: /etc/evok/hw_definitions/xS40.yaml, definition count 6
DEBUG:evok:YAML Definition loaded: /etc/evok/hw_definitions/IAQ-TH.yaml, definition count 7
DEBUG:evok:YAML Definition loaded: /etc/evok/hw_definitions/IAQ-THC.yaml, definition count 8
DEBUG:evok:YAML Definition loaded: /etc/evok/hw_definitions/01.yaml, definition count 9
DEBUG:evok:YAML Definition loaded: /etc/evok/hw_definitions/CUSTOM_MODBUS_DEVICE.yaml, definition count 10
DEBUG:evok:YAML Definition loaded: /etc/evok/hw_definitions/xS51.yaml, definition count 11
DEBUG:evok:YAML Definition loaded: /etc/evok/hw_definitions/0A.yaml, definition count 12
DEBUG:evok:YAML Definition loaded: /etc/evok/hw_definitions/xS11.yaml, definition count 13
DEBUG:evok:YAML Definition loaded: /etc/evok/hw_definitions/xS10.yaml, definition count 14
DEBUG:evok:YAML Definition loaded: /etc/evok/hw_definitions/09.yaml, definition count 15
DEBUG:evok:YAML Definition loaded: /etc/evok/hw_definitions/0F.yaml, definition count 16
DEBUG:evok:YAML Definition loaded: /etc/evok/hw_definitions/xS30.yaml, definition count 17
DEBUG:evok:Load aliases with {}
INFO:evok:HTTP server API listening on 127.0.0.1:8080
DEBUG:evok:Registered new device 'tcp_bus' with circuit LOCAL_TCP 	 (<evok.config.TcpBusDevice object at 0xb5c781f0>)
DEBUG:evok:Registered new device 'device_info' with circuit Neuron_L203_89 	 (<evok.config.DeviceInfo object at 0xb5c78e50>)
INFO:evok:Creating bus 'LOCAL_TCP' with type 'MODBUSTCP' with devices.
INFO:evok:^ Creating device '1' with type 'MODBUSTCP'
DEBUG:evok:Registered new device 'modbus_slave' with circuit 1 	 (<evok.modbus_slave.ModbusSlave object at 0xb5c78750>)
INFO:evok:^ Creating device '2' with type 'MODBUSTCP'
DEBUG:evok:Registered new device 'modbus_slave' with circuit 2 	 (<evok.modbus_slave.ModbusSlave object at 0xb5c78cf0>)
INFO:evok:^ Creating device '3' with type 'MODBUSTCP'
DEBUG:evok:Registered new device 'modbus_slave' with circuit 3 	 (<evok.modbus_slave.ModbusSlave object at 0xb5c879f0>)
DEBUG:evok:Registered new device 'run' with circuit alias 	 (<evok.devices.Aliases object at 0xb5cb6c90>)
INFO:evok:Initial reading the Modbus board on Modbus address 1	(1)
INFO:evok:Initial reading the Modbus board on Modbus address 2	(2)
INFO:evok:Initial reading the Modbus board on Modbus address 3	(3)
DEBUG:evok:Registered new device 'input' with circuit 2_01 	 (<evok.modbus_slave.Input object at 0xb5c87090>)
DEBUG:evok:Registered new device 'input' with circuit 2_02 	 (<evok.modbus_slave.Input object at 0xb5c87d10>)
DEBUG:evok:Registered new device 'input' with circuit 2_03 	 (<evok.modbus_slave.Input object at 0xb5c87f10>)
DEBUG:evok:Registered new device 'input' with circuit 2_04 	 (<evok.modbus_slave.Input object at 0xb5c87970>)
DEBUG:evok:Registered new device 'input' with circuit 2_05 	 (<evok.modbus_slave.Input object at 0xb5c76a10>)
DEBUG:evok:Registered new device 'input' with circuit 2_06 	 (<evok.modbus_slave.Input object at 0xb5c768d0>)
DEBUG:evok:Registered new device 'input' with circuit 2_07 	 (<evok.modbus_slave.Input object at 0xb5c76ab0>)
DEBUG:evok:Registered new device 'input' with circuit 2_08 	 (<evok.modbus_slave.Input object at 0xb5c76c50>)
DEBUG:evok:Registered new device 'input' with circuit 2_09 	 (<evok.modbus_slave.Input object at 0xb5c76c70>)
DEBUG:evok:Registered new device 'input' with circuit 2_10 	 (<evok.modbus_slave.Input object at 0xb5c76cd0>)
DEBUG:evok:Registered new device 'input' with circuit 2_11 	 (<evok.modbus_slave.Input object at 0xb5c76a90>)
DEBUG:evok:Registered new device 'input' with circuit 2_12 	 (<evok.modbus_slave.Input object at 0xb5c76670>)
DEBUG:evok:Registered new device 'input' with circuit 2_13 	 (<evok.modbus_slave.Input object at 0xb5c76a70>)
DEBUG:evok:Registered new device 'input' with circuit 2_14 	 (<evok.modbus_slave.Input object at 0xb5c76130>)
DEBUG:evok:Registered new device 'input' with circuit 2_15 	 (<evok.modbus_slave.Input object at 0xb5c767f0>)
DEBUG:evok:Registered new device 'input' with circuit 2_16 	 (<evok.modbus_slave.Input object at 0xb5c76570>)
DEBUG:evok:Registered new device 'relay' with circuit 2_01 	 (<evok.modbus_slave.Relay object at 0xb5c76bb0>)
DEBUG:evok:Registered new device 'relay' with circuit 2_02 	 (<evok.modbus_slave.Relay object at 0xb5c76eb0>)
DEBUG:evok:Registered new device 'relay' with circuit 2_03 	 (<evok.modbus_slave.Relay object at 0xb5c759d0>)
DEBUG:evok:Registered new device 'relay' with circuit 2_04 	 (<evok.modbus_slave.Relay object at 0xb5c75b50>)
DEBUG:evok:Registered new device 'relay' with circuit 2_05 	 (<evok.modbus_slave.Relay object at 0xb5c752d0>)
[... cut ...]
    return await operation(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/evok/lib/python3.11/site-packages/pymodbus/client/base.py", line 199, in async_execute
    resp = await asyncio.wait_for(
           ^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/tasks.py", line 479, in wait_for
    return fut.result()
           ^^^^^^^^^^^^
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Connection lost during request
WARNING:evok:Slowing down device: '2'

kratochvil01, 2024-06-20:
[11:50] Antonín Všetula
Our service department determined that the SPI bus fault was caused by the use of non-original DIN holder mounting screws. The fault was not destructive, the controller is now working properly.
 heart 1

closed 2024-06-20 by kratochvil01