UnipiTechnology/evok #139: Support for MQTT API

cewbdex, 2024-04-05:
Add support for communication via MQTT.

PEPITO82, 2024-05-20:
I'm absolute beginner when it comes to MQTT. I have just installed mosquito broker as an add-on to home assistant.
Using the ai on the edge software (URL: https://github.com/jomjol/AI-on-the-edge-device-docs) in order to digitalize my watermeter.
There I just had to enable MQTT in the settings with some details and could even activate HA discovery:

[image]

And now having the data available in HA:

[image]

A similar solution for evok would be great.

miroslav-ondra, 2026-10-11:
## Proposal of the MQTT API for Evok 3.1

The branch `dev-mqtt` (and the closed #219) is built on the structure of 3.0 and cannot be taken over to `dev-3.1`, the events, the validation and the locking of the units are elsewhere. The MQTT API will be a new implementation using the same paths as the WebSocket and the webhook.

### Architecture
- `MqttHandler` is notified about the changed devices as the webhook, the last state of each device is queued and published by a task, the changes are merged as for the WebSocket.
- A command is validated by the schemas of REST and done by `set_devices()`: under the lock of the unit, the event with the state follows the write.
- The connection to the broker is opened again after a failure, a failure is logged once.
- Library `aiomqtt` (asyncio, on paho-mqtt), a regular dependency.

### Topics (`prefix` configurable, `evok` by default)

| Topic | Direction | Payload |
|---|---|---|
| `evok/status` | Evok → | `online` / `offline`, retained, `offline` is the last will |
| `evok/<dev>/<circuit>` | Evok → | `full()` as JSON, retained, after every change and after the connection |
| `evok/<dev>/<circuit>/set` | → Evok | the params as in the body of REST, e.g. `{"value": 1, "pulse_duration": 1}`; a plain `1` / `0` / `true` is `{"value": ...}` |
| `evok/error` | Evok → | the error of a command in the format of REST: `{"topic": ..., "errors": {...}}` |

The retained states give a new client the actual state at once, no "all" command is needed. More units on one broker use their own prefix, e.g. `evok/e410-boiler`.

### Configuration (`apis.mqtt`)
```yaml
mqtt:
  enabled: false
  hostname: 127.0.0.1
  port: 1883
  username: evok          # optional
  password: secret        # optional
  tls: false              # or ca_file / cert_file / key_file
  client_id: evok-<hostname>
  prefix: evok
  qos: 0
  device_mask: [di, ro, do, ai, ao, led, sensor, wd, data_point]   # as the webhook, all by default
  read_only: false        # true: the /set topics are not subscribed
```

### Security
The access is controlled by the broker (user, password, ACL, TLS), the token of the API is not used by MQTT. `read_only: true` disables the commands. A broker without a user and the commands enabled are logged once at the start.

### Stages
1. States, `/set`, the last will, the configuration, tests, docs, a live test with mosquitto.
2. Home Assistant discovery (`homeassistant/<component>/<node>/<object>/config`: DI `binary_sensor`, RO/DO/LED `switch`, AI and 1-Wire `sensor`, AO `number`), enabled by `ha_discovery: true`.
3. Later, optional: the state of one device on request, setting by the alias.

miroslav-ondra, 2026-10-12:
Implemented in Evok 3.1 (branch `dev-3.1`), it will be closed with the release:

- **MQTT API** (#246, commit c8ff385): the states of the devices are published retained to `<prefix>/<device>/<circuit>` (`evok` by default), the commands are received by `<prefix>/<device>/<circuit>/set` (the params as in REST or a plain value), `<prefix>/status` is `online` / `offline` with the last will, the errors of the commands go to `<prefix>/error`. The access is controlled by the broker (user, password, ACL, TLS), `read_only` disables the commands.
- **Home Assistant discovery** (#247, commit 0538f51): with `ha_discovery: true` the devices are announced by the MQTT discovery — DI binary sensors, RO/DO/LED switches, AI sensors, AO numbers, data points, 1-Wire thermometers and DS2438, the communication of the Modbus units; one Home Assistant device for each Modbus unit and 1-Wire bus. Tested with Home Assistant 2026.10.1.

Configuration (`apis.mqtt` in `config.yaml`):
```yaml
apis:
  mqtt:
    enabled: true
    hostname: 192.168.1.10
    username: evok
    password: secret
    ha_discovery: true
```

Documentation: `docs/apis/mqtt.md`, the options in `docs/configs/evok_configuration.md`.

closed 2026-11-02 by matoushybl