UnipiTechnology/evok #141: evok becomes unresponsive with high rate input changes

rudyvan, 2024-04-06:
Is with the latest evok release, a RPI 4 running bookworm

I have a unipi 1.1 connected to a wind sensor from eltako on input 1_12

[image]

When there is a lot of wind, it turns rapidly, generating many input changes and therefore many websocket calls.
i made sure that the processing on the websocket server is very fast as not holding up, and i am using the counter output, not the individual input changes, so there is no issue when i miss an input change.

But after a while the evok api goes dead and does not communicate any input changes any more, not only for 1-12 but for all inputs.  Also evok-web becomes static and does not show input changes.

With the old evok application, i did not had this issue and it always worked, even with a lot of wind.

I assume some tests are necessary to monitor what happens with evok at high rate input changes as it is crucial that evok never stops working properly.

In my opinion, evok should suspend sending input changes when websocket (or other webhook) buffers are running full until they are empty again, but not die trying.
Missing input changes is not an issue as long as the counters in that case stay true as the change count can be derived.

Can this be investigated?

kratochvil01, 2024-04-07:
I'll take a look at it.

Was there anything interesting in the Evok or unipi-one log?
Do other peripherals work normally?
When does Evok recover from the state of not sending anything? Is it true that once events stop coming in on DI 1_12, Evok communicates normally again?

Thank you for your feedback.

rudyvan, 2024-04-07:
i didn't change any setting, so the debounce setting must be the default one.

where can i find these log files?

Once evok hangs, there is nothing coming out of the websocket and it does not restarts automatically after a while (the wind keeps generating input changes anyway, i did not test by disconnecting the input).

strangely enough, the command 'sudo service evok restart' followed by 'sudo service nginx restart' does not solve the issue, you have to reboot and then everything is again normal (until next freeze).
after service restart, evok-web only shows the header with no input or relay lines (only 1w sensors work). 

reboot is the only way out to have evok running again. 

hope it helps finding the cause.

miroslav-ondra, 2024-04-08:
There was a bug with counters in unipi-modbus-one. Try apt update && apt install unipi-modbus-one.  Current version 0.24.0

rudyvan, 2024-04-08:
i suppose you mean update && apt install unipi-one-modbus
it is not very windy today, but i will try with a leaf blower to simulate a hurricane :)

miroslav-ondra, 2024-04-08:
Do you know the frequency of input stream from windmeter? And how many "fast switching"  do you have connected?
I tried 1kHz.

rudyvan, 2024-04-08:
just tried with a wind blower and registered 96 km/h.  Evok hold out, so i think issue is solved.   Thanks!

closed 2024-04-08 by rudyvan