UnipiTechnology/evok #145: Unable to locate package evok error when trying to install beta

PEPITO82, 2024-04-17:
Hi,

when following the beta installation for unipi 1.1. from this URL: https://github.com/UniPiTechnology/evok/blob/dev-bookworm/docs/installation.md#installing-and-updating-evok I'm getting "Unable to locate package evok" error:

`pi@homeassistant:~ $ ```bash
echo "deb https://repo.unipi.technology/debian betaevok:bookworm main unipi1-mai                                                                                                                                                                                                                                                                                                                                                           n unipi1-test test" >> /etc/apt/sources.list.d/unipi.list
apt update
```
pi@homeassistant:~ $ sudo apt update
sudo apt install -y evok

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

W: GPG error: https://repo.unipi.technology/debian bookworm InRelease: The follo                                                                                                                                                                                                                                                                                                                                                           wing signatures couldn't be verified because the public key is not available: NO                                                                                                                                                                                                                                                                                                                                                           _PUBKEY 295F73416272C271
E: The repository 'https://repo.unipi.technology/debian bookworm InRelease' is n                                                                                                                                                                                                                                                                                                                                                           ot signed.

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

E: Unable to locate package evok
`

cewbdex, 2024-04-17:
Please run the script, not commands by hand.

kratochvil01, 2024-04-17:
You dont have Unipi repository signing key. 
Did you run the [command to add unipi repository](https://evok.readthedocs.io/en/latest/installation/#raspberrypi-os)?

cewbdex, 2024-04-17:
You did not follow the guide, you need to set up non-beta repos first.

closed 2024-04-17 by cewbdex

PEPITO82, 2024-04-17:
@cewbdex I actually did before. But I copied into console the complete commands from URL. Looks like these should be executed line by line. And without the formatting:

```bash
echo "deb https://repo.unipi.technology/debian betaevok:bookworm main unipi1-main unipi1-test test" >> /etc/apt/sources.list.d/unipi.list
apt update
```

cewbdex, 2024-04-17:
You lack the correct GPG signatures in your system, which are installed by the script:
[image]