UnipiTechnology/evok #157: [Feature]: Refactor I/O naming

tomashora, 2024-05-02:
### Describe the feature

Refactor naming of I/Os to di,do,ai,ao,ro (which should be used primarly). Obsolete /input and /output endpoints could be preserved as aliases if it brings 100% backward compatibility (so users won't have to change their app at all).

Most work will have to be done in the documentation and javascript of evok-web.

Make sure to cleanup the devtype_names and aliases in devices.py from unsupported/obsolete devices.

### Use case

Access to I/Os will match naming conventions on the products however, it will break userapplications using these API calls

kratochvil01, 2024-05-02:
Changes must be refrected in:

- [ ] evok-web.
- [ ] documentation.
- [ ] API documentation. 
- [ ] node-red nodes

kratochvil01, 2024-05-10:
https://github.com/UniPiTechnology/evok/pull/158

closed 2024-05-10 by kratochvil01