UnipiTechnology/evok #161: [Bug]: Relay 7 not showing current state in node-red

PEPITO82, 2024-05-20:
### Prerequisites

- [X] I am running the latest Evok version
- [X] I have searched other [issues](https://github.com/UniPiTechnology/evok/issues)

### Describe the bug

I have just installed the latest beta OS image for Unipi1 64-bit.
When opening node-red, the state of other relays is reflecting just fine but not for Relay 7:

[image]

No state in node red is showing for Relay 7.

### Steps to reproduce

1. Go to https://evok.readthedocs.io/en/latest/installation/ and download the latest Node-Red OS imae for Unipi1 64-bit
2. Write to SD card and run it on Raspberry
3. Install/update Evok and install the Evok-web

sudo apt install -y evok
sudo apt install -y evok-web

4. Open Node-Red in browser... IP:1880
5. Switch Relays in Evok-web and see if status is reflected correctly in Node-Red.

### Expected behavior

State of Relay 7 to be shown in Node-Red

### Actual behavior

State of Relay 7 not showing in Node-Red

PEPITO82, 2024-05-21:
This can easily be fixed by changing the property for device type to "Relay":

[image]

kratochvil01, 2024-05-21:
Hello,
The problem is the wrong device type in the example Node-red project. I have solve the error and everything will be OK in the next release of the operating system.
Thank you very much for your feedback.

closed 2024-05-22 by PEPITO82