UnipiTechnology/evok #162: [Bug]: Cannot connect via websocket Evok 3 on my unipi 1.1

PEPITO82, 2024-05-21:
### Prerequisites

- [X] I am running the latest Evok version
- [X] I have searched other [issues](https://github.com/UniPiTechnology/evok/issues)

### Describe the bug

Hi all,

I would like to connect via websocket to Evok 3 on my unipi board 1.1 by using the Node-RED instance of my home assistant installation.

Installed the latest https://kb.unipi.technology/files:software:os-images:unipi1x64-base-os-hidden on my RPI3 B+ as I thought I cannot connect via websocket if I'm running other Node-RED instance locally on RPI3 with attached unipi board 1.1.

But also after installing fresh base OS, I still cannot connect from Node-RED in my home assistant installation:
Status remains "disconnected":
[image]

I'm using the flows which came with the Node-Red image for unipi 1.1.
Changed the IP of websocket server to the one Evok is using:
[image]

Not sure if I'm doing anything wrong or this is a bug.

Thanks for your help.

### Steps to reproduce

1. Import UniPi flows to Node-RED installation.
2. Change IP from websocket-client in Node-RED.

### Expected behavior

I can connect to websocket-client which is not local to my home assistant installation

### Actual behavior

No connection possible for Node-RED instance running on home assistant server.

PEPITO82, 2024-05-22:
Hi @kratochvil01.

thanks for your help.
I proceeded with option 1) and changed the websocket-client settings in my Node-RED instance running as home assistant add-on.

Now it's connected:

[image]

One last general question on websocket. Is there only one connection via websocket possible or multiple to Evok?

Regards,
Peter

kratochvil01, 2024-05-22:
The number of connections created using the websocket protocol is not limited. However, the more there are, the more connections Evok will have to handle in case of an event. Therefore, if there are too many connections, high latency can occur.

If you plan to create more connections using the websocket protocol (in the order of tens or hundreds), we strongly recommend setting up [device filters](https://evok.readthedocs.io/en/latest/apis/websocket/#listening-on-websocket-with-filter-on-do-and-ao).