UnipiTechnology/evok #163: [Feature]: more comprehensive filter possibility with web_socket

rudyvan, 2024-06-08:
### Describe the feature

from the documentation, it is possible with web_socket to filter certain devices, but not certain circuits.

(extract)

```python
def on_open(ws):
    print("WebSocket connection opened")
    msg = {"cmd": "filter", "devices": ["do", "ao"]}
    ws.send(json.dumps(msg))
```

Would it be possible to filter certain devices/circuits?

say:
```
    msg = {"cmd": "filter", "devices": {"do": ["1_01", "1_02"], "ao": ["1_01"]}}
```

Why?

It seems that i am getting a lot of 1_12 input changes, even when the input is made high.
On some cases i have 4 web_sockets on one unipi and i would like to limit trafic to the inputs/outputs that are in use and nothing from the others.

### Use case

less network traffic for non used circuits

kratochvil01, 2024-06-13:
This is a very useful feature. But I think it might be better to group devices and then add the ability to filter based on these groups.

For example create a Kitchen group with devices like lights, switches, thermometers. etc. And then set the WS filter only for this room. What do you think?

rudyvan, 2024-06-13:
Any way of filtering a group of things is useful as unused inputs can occasionally generate a lot of traffic.
Today, when you use analog input 1_01, then sometimes the unused analog input 1_02 start to generate a lot of traffic oscillating between 0,01 and 0,0.    Could be avoided with filtering.

So your suggestion of using a group of devices to filter is perfectly fine i think.
Other solution is to unfilter devices such as the ai,1_02 when they are not used in the application.