UnipiTechnology/evok #166: [Bug]: Unipi-os-configurator-data conflict pacakges

GrzegorzWegrzyn, 2024-06-20:
### Prerequisites

- [X] I am running the latest Evok version
- [X] I have searched other [issues](https://github.com/UniPiTechnology/evok/issues)

### Describe the bug

Hi all,

i'm using UniPi L203 device with RPi OS (64bit, bookworm)

I cannot install Evok on a fresh Raspberry Pi OS. When doing "OS preparation" step i've got following conflict:

`Running Raspberry Pi OS Debian/bookworm on arm64
Downloading and importing the Unipi repository signing key

Downloading the APT definition for Unipi repository
deb https://repo.unipi.technology/debian bookworm neuron-main main

Updating the package list
Hit:1 http://archive.raspberrypi.com/debian bookworm InRelease
Hit:2 http://deb.debian.org/debian bookworm InRelease
Hit:3 http://deb.debian.org/debian-security bookworm-security InRelease
Hit:4 http://deb.debian.org/debian bookworm-updates InRelease
Get:5 https://repo.unipi.technology/debian bookworm InRelease [127 kB]
Get:6 https://repo.unipi.technology/debian bookworm/neuron-main armhf Packages [6,852 B]
Get:7 https://repo.unipi.technology/debian bookworm/neuron-main arm64 Packages [19.6 kB]
Get:8 https://repo.unipi.technology/debian bookworm/neuron-main all Packages [1,956 B]
Get:9 https://repo.unipi.technology/debian bookworm/main armhf Packages [11.8 kB]
Get:10 https://repo.unipi.technology/debian bookworm/main all Packages [6,416 B]
Get:11 https://repo.unipi.technology/debian bookworm/main arm64 Packages [12.0 kB]
Fetched 186 kB in 2s (107 kB/s)
Reading package lists... Done
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
unipi-kernel-modules is already the newest version (2.58.6.6.31-1+rpt1).
unipi-firmware-tools is already the newest version (2.20~bookworm).
unipi-modbus-tools is already the newest version (2.20~bookworm).
unipi-firmware6 is already the newest version (6.22~bookworm).
The following packages were automatically installed and are no longer required:
  libcamera0.2 libraspberrypi0 libwpe-1.0-1 libwpebackend-fdo-1.0-1
Use 'sudo apt autoremove' to remove them.
The following NEW packages will be installed:
  unipi-os-configurator-data
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
1 not fully installed or removed.
Need to get 0 B/15.7 kB of archives.
After this operation, 229 kB of additional disk space will be used.
(Reading database ... 151529 files and directories currently installed.)
Preparing to unpack .../unipi-os-configurator-data_0.68~bookworm-neuron64u_arm64.deb ...
Leaving 'diversion of /boot/overlays/ds2482.dtbo to /usr/share/rpikernelhack/overlays/ds2482.dtbo by rpikernelhack'
Leaving 'diversion of /boot/overlays/unipi_akuso1.dtbo to /usr/share/rpikernelhack/overlays/unipi_akuso1.dtbo by rpikernelhack'
Leaving 'diversion of /boot/overlays/unipi_id.dtbo to /usr/share/rpikernelhack/overlays/unipi_id.dtbo by rpikernelhack'
Leaving 'diversion of /boot/overlays/unipi_l203.dtbo to /usr/share/rpikernelhack/overlays/unipi_l203.dtbo by rpikernelhack'
Leaving 'diversion of /boot/overlays/unipi_l303.dtbo to /usr/share/rpikernelhack/overlays/unipi_l303.dtbo by rpikernelhack'
Leaving 'diversion of /boot/overlays/unipi_l403.dtbo to /usr/share/rpikernelhack/overlays/unipi_l403.dtbo by rpikernelhack'
Leaving 'diversion of /boot/overlays/unipi_l503.dtbo to /usr/share/rpikernelhack/overlays/unipi_l503.dtbo by rpikernelhack'
Leaving 'diversion of /boot/overlays/unipi_l513.dtbo to /usr/share/rpikernelhack/overlays/unipi_l513.dtbo by rpikernelhack'
Leaving 'diversion of /boot/overlays/unipi_l523.dtbo to /usr/share/rpikernelhack/overlays/unipi_l523.dtbo by rpikernelhack'
Leaving 'diversion of /boot/overlays/unipi_l533.dtbo to /usr/share/rpikernelhack/overlays/unipi_l533.dtbo by rpikernelhack'
Leaving 'diversion of /boot/overlays/unipi_m103.dtbo to /usr/share/rpikernelhack/overlays/unipi_m103.dtbo by rpikernelhack'
Leaving 'diversion of /boot/overlays/unipi_m203.dtbo to /usr/share/rpikernelhack/overlays/unipi_m203.dtbo by rpikernelhack'
Leaving 'diversion of /boot/overlays/unipi_m303.dtbo to /usr/share/rpikernelhack/overlays/unipi_m303.dtbo by rpikernelhack'
Leaving 'diversion of /boot/overlays/unipi_m403.dtbo to /usr/share/rpikernelhack/overlays/unipi_m403.dtbo by rpikernelhack'
Leaving 'diversion of /boot/overlays/unipi_m503.dtbo to /usr/share/rpikernelhack/overlays/unipi_m503.dtbo by rpikernelhack'
Leaving 'diversion of /boot/overlays/unipi_m523.dtbo to /usr/share/rpikernelhack/overlays/unipi_m523.dtbo by rpikernelhack'
Leaving 'diversion of /boot/overlays/unipi_rtc.dtbo to /usr/share/rpikernelhack/overlays/unipi_rtc.dtbo by rpikernelhack'
Leaving 'diversion of /boot/overlays/unipi_s103.dtbo to /usr/share/rpikernelhack/overlays/unipi_s103.dtbo by rpikernelhack'
Leaving 'diversion of /boot/overlays/unipi_s103_e.dtbo to /usr/share/rpikernelhack/overlays/unipi_s103_e.dtbo by rpikernelhack'
Leaving 'diversion of /boot/overlays/unipi_s103_g.dtbo to /usr/share/rpikernelhack/overlays/unipi_s103_g.dtbo by rpikernelhack'
Leaving 'diversion of /boot/overlays/unipi_s103_i.dtbo to /usr/share/rpikernelhack/overlays/unipi_s103_i.dtbo by rpikernelhack'
Unpacking unipi-os-configurator-data (0.68~bookworm-neuron64u) ...
dpkg: error processing archive /var/cache/apt/archives/unipi-os-configurator-data_0.68~bookworm-neuron64u_arm64.deb (--unpack):
 trying to overwrite '/lib/systemd/system/sshswitch.service', which is also in package raspberrypi-sys-mods 20240619
Errors were encountered while processing:
 /var/cache/apt/archives/unipi-os-configurator-data_0.68~bookworm-neuron64u_arm64.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
Cleaning temporary files
--------------------------------------------------------------
Reboot is required to load the kernel module and so run reboot
--------------------------------------------------------------
Cleaning temporary files`

I tried to turn off the SSH service vias raspi-config, but it didn't help

### Steps to reproduce

1. Flash new image using latest RPi OS (64bit)
2. Install HomeAssistant Core
3. Try to "prepare OS" for Evok installation

### Expected behavior

System should be prepared for Evok installation

### Actual behavior

Package conflict

kratochvil01, 2024-06-24:
Hello,
We've resolved the conflict. Everything should be working fine now. Please run the `apt update` command and then run the installation script again.

kratochvil01, 2024-06-24:
Hello,
We've resolved the conflict. Everything should be working fine now. Please run the installation script again.

PEPITO82, 2024-06-24:
@kratochvil01 I have just tried to bring system ([unipi1x64-base-os_12]) to latest status with following commands:

sudo apt update
sudo apt full-upgrade

But I'm also getting the same error and upgrade is stopped:

[image]

kratochvil01, 2024-06-24:
In the old version of our package there was this error. A new version has been released. Please try calling `apt update` before running this command.
If the error persists, try upgrading our package with `apt install unipi-os-configurator-data` first and call full-upgrade later.

kratochvil01, 2024-06-24:
Hmm. I see that the bug I wrote about can occur with the Neuron platform. Your listing looks like a problem between `raspberrypi-sys-mods` and `libpython3.11`.

kratochvil01, 2024-06-25:
The error is these conflicting packages are accessing the same file. Normally these files are diverted, but in our OS build this will not happen. We will fix this bug in the next OS release.
For this error, please call the command: 
`dpkg-divert --package raspberrypi-sys-mods --add --rename --divert /usr/lib/python3.11/EXTERNALLY-MANAGED.orig /usr/lib/python3.11/EXTERNALLY-MANAGED`

kratochvil01, 2024-07-01:
@GrzegorzWegrzyn 
Was the installation completed correctly?

PEPITO82, 2024-07-01:
@kratochvil01 Thanks for your help. With help of the dpkg-divert command you shared, I finally could do sudo apt full-upgrade.

GrzegorzWegrzyn, 2024-07-04:
@kratochvil01 
Sorry for the delay. Yes, the installation was successful. Thanks a lot

closed 2024-07-04 by GrzegorzWegrzyn