UnipiTechnology/evok #187: [Bug]: 1wire no devide showing up!

pgoenczi, 2024-09-19:
### Prerequisites

- [X] I am running the latest Evok version
- [X] I have searched other [issues](https://github.com/UniPiTechnology/evok/issues)

### Describe the bug

HW: Unipi1.1 / Raspberry Pi 3B+ / 1 Wire HUB / DS18B20 / DS2438 (sensors & hub are official Unipi)

SW: I had an earlier version of Evok / Node-Red / Raspi OS (bullseye) perfetly working with teh other setup

**I have fresh installed the latest SW ( Evok / Node-Red / Raspi OS)  -> no 1wire devices show up on the evok-web**

It has to be a config issue I guess, but I cant sort it out.

### Evok log

Sep 18 20:18:07 UNIPI1-sn-1787822080 systemd[1]: Started evok.service - EVOK Rest/WebSocket/RPC Server.
Sep 18 20:18:18 UNIPI1-sn-1787822080 evok[730]: INFO:evok:Starting Evok v3.0.6 using config directory '/etc/evok'.
Sep 18 20:18:18 UNIPI1-sn-1787822080 evok[730]: INFO:evok:Setting logging level to 'WARNING'.
Sep 18 20:20:44 UNIPI1-sn-1787822080 systemd[1]: Stopping evok.service - EVOK Rest/WebSocket/RPC Server...
Sep 18 20:20:45 UNIPI1-sn-1787822080 systemd[1]: evok.service: Deactivated successfully.
Sep 18 20:20:45 UNIPI1-sn-1787822080 systemd[1]: Stopped evok.service - EVOK Rest/WebSocket/RPC Server.
Sep 18 20:20:45 UNIPI1-sn-1787822080 systemd[1]: evok.service: Consumed 29.535s CPU time.
Sep 18 20:20:45 UNIPI1-sn-1787822080 systemd[1]: Started evok.service - EVOK Rest/WebSocket/RPC Server.
Sep 18 20:20:49 UNIPI1-sn-1787822080 evok[2077]: INFO:evok:Starting Evok v3.0.6 using config directory '/etc/evok'.
Sep 18 20:20:49 UNIPI1-sn-1787822080 evok[2077]: INFO:evok:Setting logging level to 'WARNING'.
Sep 19 10:06:14 UNIPI1-sn-1787822080 systemd[1]: Stopping evok.service - EVOK Rest/WebSocket/RPC Server...
Sep 19 10:06:15 UNIPI1-sn-1787822080 systemd[1]: evok.service: Deactivated successfully.
Sep 19 10:06:15 UNIPI1-sn-1787822080 systemd[1]: Stopped evok.service - EVOK Rest/WebSocket/RPC Server.
Sep 19 10:06:15 UNIPI1-sn-1787822080 systemd[1]: evok.service: Consumed 2h 14min 29.372s CPU time.
Sep 19 10:06:15 UNIPI1-sn-1787822080 systemd[1]: Started evok.service - EVOK Rest/WebSocket/RPC Server.
Sep 19 10:06:18 UNIPI1-sn-1787822080 evok[3212]: INFO:evok:Starting Evok v3.0.6 using config directory '/etc/evok'.
Sep 19 10:06:18 UNIPI1-sn-1787822080 evok[3212]: INFO:evok:Setting logging level to 'DEBUG'.
Sep 19 10:06:18 UNIPI1-sn-1787822080 evok[3212]: DEBUG:evok:YAML Definition loaded: /etc/evok/hw_definitions/xS11.yaml, definition count 0
Sep 19 10:06:18 UNIPI1-sn-1787822080 evok[3212]: DEBUG:evok:YAML Definition loaded: /etc/evok/hw_definitions/xS30.yaml, definition count 1
Sep 19 10:06:18 UNIPI1-sn-1787822080 evok[3212]: DEBUG:evok:YAML Definition loaded: /etc/evok/hw_definitions/IAQ-THC.yaml, definition count 2
Sep 19 10:06:18 UNIPI1-sn-1787822080 evok[3212]: DEBUG:evok:YAML Definition loaded: /etc/evok/hw_definitions/xG18.yaml, definition count 3
Sep 19 10:06:18 UNIPI1-sn-1787822080 evok[3212]: DEBUG:evok:YAML Definition loaded: /etc/evok/hw_definitions/xS50.yaml, definition count 4
Sep 19 10:06:18 UNIPI1-sn-1787822080 evok[3212]: DEBUG:evok:YAML Definition loaded: /etc/evok/hw_definitions/xS51.yaml, definition count 5
Sep 19 10:06:18 UNIPI1-sn-1787822080 evok[3212]: DEBUG:evok:YAML Definition loaded: /etc/evok/hw_definitions/UNIPI11.yaml, definition count 6
Sep 19 10:06:18 UNIPI1-sn-1787822080 evok[3212]: DEBUG:evok:YAML Definition loaded: /etc/evok/hw_definitions/xS10.yaml, definition count 7
Sep 19 10:06:18 UNIPI1-sn-1787822080 evok[3212]: DEBUG:evok:YAML Definition loaded: /etc/evok/hw_definitions/IAQ-TH.yaml, definition count 8
Sep 19 10:06:18 UNIPI1-sn-1787822080 evok[3212]: DEBUG:evok:YAML Definition loaded: /etc/evok/hw_definitions/EMO-R8.yaml, definition count 9
Sep 19 10:06:18 UNIPI1-sn-1787822080 evok[3212]: DEBUG:evok:YAML Definition loaded: /etc/evok/hw_definitions/CUSTOM_MODBUS_DEVICE.yaml, definition count 10
Sep 19 10:06:18 UNIPI1-sn-1787822080 evok[3212]: DEBUG:evok:YAML Definition loaded: /etc/evok/hw_definitions/xS40.yaml, definition count 11
Sep 19 10:06:19 UNIPI1-sn-1787822080 evok[3212]: DEBUG:evok:YAML Definition loaded: /etc/evok/hw_definitions/UNIPI11LITE.yaml, definition count 12
Sep 19 10:06:19 UNIPI1-sn-1787822080 evok[3212]: DEBUG:evok:Load aliases with {}
Sep 19 10:06:19 UNIPI1-sn-1787822080 evok[3212]: INFO:evok:HTTP server API listening on 127.0.0.1:8080
Sep 19 10:06:19 UNIPI1-sn-1787822080 evok[3212]: DEBUG:evok:Registered new device 'tcp_bus' with circuit LOCAL_TCP          (<evok.config.TcpBusDevice object at 0x7f87a07890>)
Sep 19 10:06:19 UNIPI1-sn-1787822080 evok[3212]: DEBUG:evok:Registered new device 'device_info' with circuit UNIPI11          (<evok.config.DeviceInfo object at 0x7f87a06990>)
Sep 19 10:06:19 UNIPI1-sn-1787822080 evok[3212]: INFO:evok:Creating bus 'LOCAL_TCP' with type 'MODBUSTCP' with devices.
Sep 19 10:06:19 UNIPI1-sn-1787822080 evok[3212]: INFO:evok:^ Creating device '1' with type 'MODBUSTCP'
Sep 19 10:06:19 UNIPI1-sn-1787822080 evok[3212]: DEBUG:evok:Registered new device 'modbus_slave' with circuit 1          (<evok.modbus_slave.ModbusSlave object at 0x7f87a07110>)
Sep 19 10:06:19 UNIPI1-sn-1787822080 evok[3212]: INFO:evok:Creating bus 'OWFS' with type 'OWBUS'.
Sep 19 10:06:19 UNIPI1-sn-1787822080 evok[3212]: DEBUG:evok:Registered new device 'run' with circuit alias          (<evok.devices.Aliases object at 0x7f879ef090>)
Sep 19 10:06:19 UNIPI1-sn-1787822080 evok[3212]: INFO:evok:Initial reading the Modbus board on Modbus address 0        (1)
Sep 19 10:06:19 UNIPI1-sn-1787822080 evok[3212]: DEBUG:evok:Registered new device 'ao' with circuit 1_01          (<evok.modbus_slave.AnalogOutput object at 0x7f87a077d0>)
Sep 19 10:06:19 UNIPI1-sn-1787822080 evok[3212]: DEBUG:evok:Registered new device 'ai' with circuit 1_01          (<evok.modbus_slave.AnalogInput object at 0x7f879f9790>)
Sep 19 10:06:19 UNIPI1-sn-1787822080 evok[3212]: DEBUG:evok:Registered new device 'ai' with circuit 1_02          (<evok.modbus_slave.AnalogInput object at 0x7f879f8a90>)
Sep 19 10:06:19 UNIPI1-sn-1787822080 evok[3212]: DEBUG:evok:Registered new device 'ro' with circuit 1_01          (<evok.modbus_slave.Relay object at 0x7f879f9890>)
Sep 19 10:06:19 UNIPI1-sn-1787822080 evok[3212]: DEBUG:evok:Registered new device 'ro' with circuit 1_02          (<evok.modbus_slave.Relay object at 0x7f879fbe10>)
Sep 19 10:06:19 UNIPI1-sn-1787822080 evok[3212]: DEBUG:evok:Registered new device 'ro' with circuit 1_03          (<evok.modbus_slave.Relay object at 0x7f879f9210>)
Sep 19 10:06:19 UNIPI1-sn-1787822080 evok[3212]: DEBUG:evok:Registered new device 'ro' with circuit 1_04          (<evok.modbus_slave.Relay object at 0x7f879fa590>)
Sep 19 10:06:19 UNIPI1-sn-1787822080 evok[3212]: DEBUG:evok:Registered new device 'ro' with circuit 1_05          (<evok.modbus_slave.Relay object at 0x7f879fa4d0>)
Sep 19 10:06:19 UNIPI1-sn-1787822080 evok[3212]: DEBUG:evok:Registered new device 'ro' with circuit 1_06          (<evok.modbus_slave.Relay object at 0x7f879f91d0>)
Sep 19 10:06:19 UNIPI1-sn-1787822080 evok[3212]: DEBUG:evok:Registered new device 'ro' with circuit 1_07          (<evok.modbus_slave.Relay object at 0x7f879faf90>)
Sep 19 10:06:19 UNIPI1-sn-1787822080 evok[3212]: DEBUG:evok:Registered new device 'ro' with circuit 1_08          (<evok.modbus_slave.Relay object at 0x7f879f8cd0>)
Sep 19 10:06:19 UNIPI1-sn-1787822080 evok[3212]: DEBUG:evok:Registered new device 'di' with circuit 1_01          (<evok.modbus_slave.DigitalInput object at 0x7f879fa810>)
Sep 19 10:06:19 UNIPI1-sn-1787822080 evok[3212]: DEBUG:evok:Registered new device 'di' with circuit 1_02          (<evok.modbus_slave.DigitalInput object at 0x7f879f82d0>)
[... 3 lines cut ...]
Sep 19 10:06:19 UNIPI1-sn-1787822080 evok[3212]: DEBUG:evok:Registered new device 'di' with circuit 1_06          (<evok.modbus_slave.DigitalInput object at 0x7f879fa910>)
Sep 19 10:06:19 UNIPI1-sn-1787822080 evok[3212]: DEBUG:evok:Registered new device 'di' with circuit 1_07          (<evok.modbus_slave.DigitalInput object at 0x7f879fb3d0>)
Sep 19 10:06:19 UNIPI1-sn-1787822080 evok[3212]: DEBUG:evok:Registered new device 'di' with circuit 1_08          (<evok.modbus_slave.DigitalInput object at 0x7f879fb390>)
Sep 19 10:06:19 UNIPI1-sn-1787822080 evok[3212]: DEBUG:evok:Registered new device 'di' with circuit 1_09          (<evok.modbus_slave.DigitalInput object at 0x7f879fa550>)
[... cut ...]
Sep 19 10:15:00 UNIPI1-sn-1787822080 evok[3212]: INFO:evok:Shutting down
Sep 19 10:15:01 UNIPI1-sn-1787822080 systemd[1]: evok.service: Deactivated successfully.
Sep 19 10:15:01 UNIPI1-sn-1787822080 systemd[1]: Stopped evok.service - EVOK Rest/WebSocket/RPC Server.
Sep 19 10:15:01 UNIPI1-sn-1787822080 systemd[1]: evok.service: Consumed 1min 28.783s CPU time.
Sep 19 10:15:01 UNIPI1-sn-1787822080 systemd[1]: Started evok.service - EVOK Rest/WebSocket/RPC Server.

### Expected behavior

1 wire devices working

### Actual behavior

no 1 wire device work

cleveHEX, 2024-09-19:
Hello,
a temporary solution should be changing `/dev/i2c-2` to `/dev/i2c-1`, you can see if it works for you, if so, it will be resolved in future for future installs.

pgoenczi, 2024-09-25:
Do I get it right that there is no solution to make it work?

pgoenczi, 2024-09-25:
Would a "DS2484 I2C to 1-Wire Bus" do the trick?

cleveHEX, 2024-09-26:
> Do I get it right that there is no solution to make it work?
> 
> I need to rework the whole sensor group I'm running?

The solution is as I stated, edit that line in OWFS config, should be like /etc/owfs.conf.

pgoenczi, 2024-09-26:
> The solution is as I stated, edit that line in OWFS config, should be like /etc/owfs.conf.

Thank you for it solved it!
...and sorry I didn't get it for the first go. ;)

closed 2024-10-25 by cleveHEX