UnipiTechnology/evok #198: [Bug]: EMO-R8 Detection Issue

jardahrazdera, 2024-12-29:
### Prerequisites

- [X] I am running the latest Evok version
- [X] I have searched other [issues](https://github.com/UniPiTechnology/evok/issues)

### Describe the bug

# Support Request: EMO-R8 Detection Issue

## Issue Description
I am using UniPi 1.1 (serial number 7128, product ID 0x101) with EVOK running. An EMO-R8 expansion board is connected via the I²C bus.

The EMO-R8 board is located at address `0x21` on the I²C bus, confirmed with the `i2cdetect` command. The physical connection and power supply to the board have been verified.

Although the EMO-R8 board is visible on the I²C bus, it is not automatically detected during EVOK startup or when using the `/opt/unipi/tools/os-configurator -f` tool. The EVOK logs and YAML configuration do not mention the EMO-R8 board.

The issue with detection appear  after last update of EVOK and OS. Exactly at this time EMO-R8 board stopped working. I'm using fresh installation of Base OS image from Unipi Technology web: https://kb.unipi.technology/_media/files:software:os-images:unipi1x64-base-os_12.20241210.0.zip

---

## Steps Taken to Resolve
1. Verified the board's presence on the I²C bus using `i2cdetect`:
   ```bash
   sudo i2cdetect -y 1
   ```
   The output confirms that the device is present at address `0x21`.

2. Ran `/opt/unipi/tools/os-configurator -f`, which detects UniPi 1.1 but does not recognize the connected EMO-R8 board.

3. Restarted EVOK and manually adjusted the `autogen.yaml` configuration file, but the board was still not detected.

4. Tested direct access to the I²C registers using `i2cset` and `i2cget`, which confirm that the device at the address responds, but EVOK still fails to detect it.

---

## Support Request
- I need assistance with diagnosing and configuring EVOK so that the EMO-R8 board is properly detected and functional.

---

## Additional Information
- **UniPi Product Name**: UNIPI11
- **UniPi Serial Number**: 7128
- **I²C Address**: 0x21
- **EVOK Version**: (Specify the version if known)
- **OS Configuration Output**:
  ```bash
  sudo /opt/unipi/tools/os-configurator -f

  DT='unipi_unipi11'
  UDEV='unipi_unipi11'
  HAS_DS2482='1'
  HAS_RTC='1'
  USE_ETC_MODULES='1'
  UNIPI_PLATFORM='0101'
  UNIPI_PRODUCT_ID='0101'
  UNIPI_PRODUCT_NAME='UNIPI11'
  UNIPI_PRODUCT_SERIAL='7128'
  CARDS=''
  run-parts: executing /opt/unipi/os-configurator/run.d/10-config_txt.sh
  run-parts: executing /opt/unipi/os-configurator/run.d/20-overlays.sh
  run-parts: executing /opt/unipi/os-configurator/run.d/20-uoverlays.sh
  run-parts: executing /opt/unipi/os-configurator/run.d/30-udev.sh
  run-parts: executing /opt/unipi/os-configurator/run.d/30-udevvar.sh
  run-parts: executing /opt/unipi/os-configurator/run.d/50-configure-one.sh
  run-parts: executing /opt/unipi/os-configurator/run.d/50-unipi-hostname.sh
  run-parts: executing /opt/unipi/os-configurator/run.d/60-evok-autogen.sh
  Detect device UNIPI1 UNIPI11 (id=0x101, sn=7128) with boards ['UNIPI11']
  run-parts: executing /opt/unipi/os-configurator/run.d/98-bootcmd.sh
  run-parts: executing /opt/unipi/os-configurator/run.d/99-wait-for-resize.sh
[... 125 lines cut ...]
---

## 9. Check Available Circuits via API
- Use `curl` to query available circuits:
  ```bash
  curl --request GET \
  --url http://localhost:8080/json/all \
  --header 'Accept: application/json'
  ```

---

## 10. Use the OS Configurator Tool
- Run the UniPi configurator to detect and configure devices:
  ```bash
  sudo /opt/unipi/tools/os-configurator -f
  ```

### Expected behavior

Working EMO-R8.

### Actual behavior

EMO-R8 is not working as expected.

cleveHEX, 2025-01-02:
Unipi 1.1 boards do not have an automatic detection, please refer to `/etc/unipi-id/README.md` file for instructions.

jardahrazdera, 2025-01-02:
I’m sorry. I made a mistake and forgot this important information. It has been a few months since I last configured Unipi and EMO, so I completely forgot this step. Is it possible that a new update of Evok deleted or overwrote some of the configuration files?

Anyway, it was my mistake. At least it’s a lesson for others in the future. Thank you for helping me refresh my memory.

jardahrazdera, 2025-01-02:
Ok. Thanks for all the information. I'm closing this issue, because it's solved.

closed 2025-01-08 by cleveHEX