UnipiTechnology/evok #199: [Bug]: High idle CPU usage

n9, 2025-01-03:
### Prerequisites

- [X] I am running the latest Evok version
- [X] I have searched other [issues](https://github.com/UniPiTechnology/evok/issues)

### Describe the bug

Evok process eats CPU when idle.

### Evok log

INFO:evok:Starting Evok v3.0.6 using config directory '/etc/evok'.
INFO:evok:Setting logging level to 'DEBUG'.
DEBUG:evok:YAML Definition loaded: /etc/evok/hw_definitions/xS51.yaml, definition count 0
DEBUG:evok:YAML Definition loaded: /etc/evok/hw_definitions/xG18.yaml, definition count 1
DEBUG:evok:YAML Definition loaded: /etc/evok/hw_definitions/xS40.yaml, definition count 2
DEBUG:evok:YAML Definition loaded: /etc/evok/hw_definitions/IAQ-THC.yaml, definition count 3
DEBUG:evok:YAML Definition loaded: /etc/evok/hw_definitions/EMO-R8.yaml, definition count 4
DEBUG:evok:YAML Definition loaded: /etc/evok/hw_definitions/xS11.yaml, definition count 5
DEBUG:evok:YAML Definition loaded: /etc/evok/hw_definitions/xS50.yaml, definition count 6
DEBUG:evok:YAML Definition loaded: /etc/evok/hw_definitions/xS30.yaml, definition count 7
DEBUG:evok:YAML Definition loaded: /etc/evok/hw_definitions/UNIPI11LITE.yaml, definition count 8
DEBUG:evok:YAML Definition loaded: /etc/evok/hw_definitions/xS10.yaml, definition count 9
DEBUG:evok:YAML Definition loaded: /etc/evok/hw_definitions/IAQ-TH.yaml, definition count 10
DEBUG:evok:YAML Definition loaded: /etc/evok/hw_definitions/CUSTOM_MODBUS_DEVICE.yaml, definition count 11
DEBUG:evok:YAML Definition loaded: /etc/evok/hw_definitions/UNIPI11.yaml, definition count 12
DEBUG:evok:Load aliases with {}
INFO:evok:HTTP server API listening on 127.0.0.1:8080
DEBUG:evok:Registered new device 'tcp_bus' with circuit LOCAL_TCP        (<evok.config.TcpBusDevice object at 0x7577c770>)
DEBUG:evok:Registered new device 'device_info' with circuit UNIPI11      (<evok.config.DeviceInfo object at 0x7577cdf0>)
INFO:evok:Creating bus 'LOCAL_TCP' with type 'MODBUSTCP' with devices.
INFO:evok:^ Creating device '1' with type 'MODBUSTCP'
DEBUG:evok:Registered new device 'modbus_slave' with circuit 1   (<evok.modbus_slave.ModbusSlave object at 0x7577c490>)
DEBUG:evok:Registered new device 'owbus' with circuit OWBUS      (<evok.owdevice.OwBusDriver object at 0x7577c4b0>)
INFO:evok:Creating bus 'OWBUS' with type 'OWFS'.
DEBUG:evok:Registered new device 'run' with circuit alias        (<evok.devices.Aliases object at 0x757b7bd0>)
INFO:evok:Initial reading the Modbus board on Modbus address 0  (1)
INFO:evok:ServerRegistered(server=<Server:127.0.0.1:4304 closed>)
DEBUG:evok:Registered new device 'ao' with circuit 1_01          (<evok.modbus_slave.AnalogOutput object at 0x75775db0>)
DEBUG:evok:Registered new device 'ai' with circuit 1_01          (<evok.modbus_slave.AnalogInput object at 0x757850d0>)
DEBUG:evok:Registered new device 'ai' with circuit 1_02          (<evok.modbus_slave.AnalogInput object at 0x75785910>)
DEBUG:evok:Registered new device 'ro' with circuit 1_01          (<evok.modbus_slave.Relay object at 0x75785830>)
DEBUG:evok:Registered new device 'ro' with circuit 1_02          (<evok.modbus_slave.Relay object at 0x757859f0>)
DEBUG:evok:Registered new device 'ro' with circuit 1_03          (<evok.modbus_slave.Relay object at 0x75785a90>)
DEBUG:evok:Registered new device 'ro' with circuit 1_04          (<evok.modbus_slave.Relay object at 0x75785ab0>)
DEBUG:evok:Registered new device 'ro' with circuit 1_05          (<evok.modbus_slave.Relay object at 0x75785b30>)
DEBUG:evok:Registered new device 'ro' with circuit 1_06          (<evok.modbus_slave.Relay object at 0x75785b90>)
DEBUG:evok:Registered new device 'ro' with circuit 1_07          (<evok.modbus_slave.Relay object at 0x75785af0>)
DEBUG:evok:Registered new device 'ro' with circuit 1_08          (<evok.modbus_slave.Relay object at 0x75785bf0>)
DEBUG:evok:Registered new device 'di' with circuit 1_01          (<evok.modbus_slave.DigitalInput object at 0x75785c50>)
DEBUG:evok:Registered new device 'di' with circuit 1_02          (<evok.modbus_slave.DigitalInput object at 0x75785c70>)
DEBUG:evok:Registered new device 'di' with circuit 1_03          (<evok.modbus_slave.DigitalInput object at 0x75785cb0>)
DEBUG:evok:Registered new device 'di' with circuit 1_04          (<evok.modbus_slave.DigitalInput object at 0x75785cf0>)
DEBUG:evok:Registered new device 'di' with circuit 1_05          (<evok.modbus_slave.DigitalInput object at 0x75785d30>)
DEBUG:evok:Registered new device 'di' with circuit 1_06          (<evok.modbus_slave.DigitalInput object at 0x75785d70>)
DEBUG:evok:Registered new device 'di' with circuit 1_07          (<evok.modbus_slave.DigitalInput object at 0x75785db0>)
DEBUG:evok:Registered new device 'di' with circuit 1_08          (<evok.modbus_slave.DigitalInput object at 0x75785df0>)
DEBUG:evok:Registered new device 'di' with circuit 1_09          (<evok.modbus_slave.DigitalInput object at 0x75785e30>)
DEBUG:evok:Registered new device 'di' with circuit 1_10          (<evok.modbus_slave.DigitalInput object at 0x75785e70>)
DEBUG:evok:Registered new device 'di' with circuit 1_11          (<evok.modbus_slave.DigitalInput object at 0x75785eb0>)
DEBUG:evok:Registered new device 'di' with circuit 1_12          (<evok.modbus_slave.DigitalInput object at 0x75785ef0>)
INFO:evok:ServerConnected(server=<Server:127.0.0.1:4304 OK>)
INFO:evok:BusAdded(bus=<Bus:<Server:127.0.0.1:4304 OK> /bus.2>)
INFO:evok:DeviceAdded(device=<TemperatureBDevice:28.AABC8E381401.5D @ None>)
INFO:evok:DeviceLocated(device=<TemperatureBDevice:28.AABC8E381401.5D @ <Bus:<Server:127.0.0.1:4304 OK> /bus.2>>)
DEBUG:evok:Registered new device 'sensor' with circuit 28AABC8E3814015D          (<evok.owdevice.DS18B20 object at 0x756d37f0>)
INFO:evok:BusAdded(bus=<Bus:<Server:127.0.0.1:4304 OK> /bus.1>)

### Steps to reproduce

1. run Evok
2. check CPU usage of the "Evok" (python) process

### Expected behavior

CPU usage between 0 - 1% (on RPI3 B+) when idle.

### Actual behavior

Something (busy waiting?) is eating CPU. CPU usage is ~16.6% (on RPI3 B+) when idle.

cleveHEX, 2025-01-07:
Evok is never `idle`, it always reads the data as defined in the config file. If you wish the cpu usage to be lower, you may lower the `scan_frequency` in the [configuration](https://evok.readthedocs.io/en/stable/configs/evok_configuration/#modbustcp-modbusrtu)

n9, 2025-01-07:
@cleveHEX Is it possible to read data only on-demand (when API requests them)?

cleveHEX, 2025-01-07:
At the moment no and it is not currently planned to be implemented. What is your use case for such need?

n9, 2025-01-07:
I assumed that Evok is a simple server that makes HW (in my case UniPI) accessible via a simple API.

But according to the configuration, it seems to me that it provides central access to various (even remote) devices (e.g. via Modbus-TCP). So the UniPI board is not accessed directly, but also via a local MODBUS server (https://github.com/UniPiTechnology/unipi-one-modbus).

In my case, direct access to the HW is preferable:
- I use onewire to read temperature sensors with longer periods (from a few seconds to a few minutes).
- And I'll use digital inputs for "accurate" monitoring of slowly switching states. (Maybe using interrupts?)

In both cases, it will be necessary to retrieve the current value if it is requested by the user.

The strange thing is that /run/unipi-plc remained even after uninstalling Evok (and rebooting). What package installed this? And how to cleanly uninstall it?

cleveHEX, 2025-01-08:
Yes, as stated in the documentation, it enables using its available API to manipulate with Modbus and 1-Wire devices. However the Modbus on Unipi 1 series is also translated, not native, they are exposed as standard Linux IO.

> I use onewire to read temperature sensors with longer periods (from a few seconds to a few minutes).

You can remove the 1-Wire sensors from Evok configuration or you can lower the polling rate significantly.

> And I'll use digital inputs for "accurate" monitoring of slowly switching states. (Maybe using interrupts?)

This is up to your implementation, use Evok, Modbus or sysfs.

> The strange thing is that /run/unipi-plc remained even after uninstalling Evok (and rebooting). What package installed this? And how to cleanly uninstall it?

This is handled by a kernel module.

n9, 2025-01-08:
> You can remove the 1-Wire sensors from Evok configuration or you can lower the polling rate significantly.

I have found that 1-Wire sensor is also available via  `/sys/bus/w1/devices/`.

> This is up to your implementation, use Evok, Modbus or sysfs.

Do any of the implementations support change notifications (via interrupts)? Or they only support polling?

> This is handled by a kernel module.

But which Debian package installed the kernel modules?

```
$ lsmod | grep unipi
unipi_id               20480  0
rtc_unipi              12288  0
regmap_i2c             12288  3 pinctrl_mcp23s08_i2c,at24,rtc_unipi
```

I thought that it was installed by the `evok` package (`apt install evok`). However, `apt remove evok` does not uninstall the kernel modules. 

```
$ dpkg-query --show --showformat '${Package} ${Status}\n' evok
evok unknown ok not-installed
```

Btw. strange is that:

```
$ cat /run/unipi-plc/unipi-id/platform_id
0101
```

but:

```
$ ls /run/unipi-plc/by-sys/
AI  DI  RO
```

```
$ cat /run/unipi-plc/by-sys/AI/in_voltage0_raw
0
```

```
$ ls /run/unipi-plc/by-sys/DI
gpiochip
```

```
ls /run/unipi-plc/by-sys/RO
gpiochip
```

There are no `/run/unipi-plc/by-sys/DI[1-3].[0-9][0-9]/value` as stated in https://kb.unipi.technology/en:sw:02-apis:04-sysfs.

cleveHEX, 2025-01-08:
> > You can remove the 1-Wire sensors from Evok configuration or you can lower the polling rate significantly.
> 
> I have found that 1-Wire sensor is also available via `/sys/bus/w1/devices/`.

Yes, that is handled by a kernel driver, if present.

> > This is up to your implementation, use Evok, Modbus or sysfs.
> 
> Do any of the implementations support change notifications (via interrupts)? Or they only support polling?

I am did never investigate this use case.
 
> > This is handled by a kernel module.
> 
> But which Debian package installed the kernel modules?
> 
> ```
> $ lsmod | grep unipi
> unipi_id               20480  0
> rtc_unipi              12288  0
> regmap_i2c             12288  3 pinctrl_mcp23s08_i2c,at24,rtc_unipi
> ```

Unipi-id, but do not remove.

> I thought that it was installed by the `evok` package (`apt install evok`). However, `apt remove evok` does not uninstall the kernel modules.

Evok has nothing to do with kernel, we do not provide support for altering kernel modules.

> ```
> $ dpkg-query --show --showformat '${Package} ${Status}\n' evok
> evok unknown ok not-installed
> ```
>
> Btw. strange is that:
>
> ```
> $ cat /run/unipi-plc/unipi-id/platform_id
> 0101
> ```
>
> but:
>
> ```
> $ ls [...]

This is applicable only to unipi PLCs (/run/unipi-plc/ "sysfs"), not boards (Unipi 1 family).

n9, 2025-01-08:
> Evok has nothing to do with kernel, we do not provide support for altering kernel modules.

After inspecting the current version of https://repo.unipi.technology/debian/raspberry-unipi1.sh (from [the documentation to install evok](https://evok.readthedocs.io/en/stable/installation/#__tabbed_1_2)), I found that the kernel modules were installed by the script (package `unipi-kernel-modules` looks supicious 🙂):

```
...

# if the release is stretch, it is probably a Neuron running older Raspbian
# prefer neuron-kernel package over unipi-kernel-modules
if [ "$(lsb_release -sr)" -ge "12" ] ; then
        PACKAGES="unipi-os-configurator-data unipi-kernel-modules unipi-one-modbus"
        ${SUDO} apt-get install -y $PACKAGES
fi

# clean temporary files
cleanup

if [ "$(lsb_release -sr)" -ge "12" ] ; then
    echo "--------------------------------------------------------------"
    echo "Reboot is required to load the kernel module and so run reboot"
    echo "--------------------------------------------------------------"
fi
# reboot

...
```
----

> This is applicable only to unipi PLCs (/run/unipi-plc/ "sysfs"), not boards (Unipi 1 family).

So only `/run/unipi-plc/unipi-id/` and `/run/unipi-plc/by-sys/AI/` is supported on UniPI 1, but `DI` and `RO` not?

----
> Unipi-id, but do not remove.

Why should I not remove `unipi-id`? (I understand that `rtc_unipi` is required for RTC.)

Can I just access DI via [System.Device.Gpio](https://learn.microsoft.com/en-us/dotnet/api/system.device.gpio) (and use [ValueChanged](https://learn.microsoft.com/en-us/dotnet/api/system.device.gpio.gpiopin.valuechanged) to receive change notification)? And RO via [Mcp23008 class](https://learn.microsoft.com/en-us/dotnet/api/iot.device.mcp23xxx.mcp23008)?

cleveHEX, 2025-01-08:
> After inspecting the current version of https://repo.unipi.technology/debian/raspberry-unipi1.sh (from [the documentation to install [...]

There is nothing suspicious at all, our hw needs specific kernel modules and packages, as every other hardware, the same as does nvidia or intel. Also the script is not obfuscated, anyone is free to see what exactly it does.

> ```
> ...
>
> # if the release is stretch, it is probably a Neuron running older Raspbian
> # prefer neuron-kernel package over unipi-kernel-modules
> if [ "$(lsb_release -sr)" -ge "12" ] ; then [...]

We do not provide any support for accessing IO by /run/unipi-plc sysfs, that is why there is no unipi 1 series tag on the page.

> Why should I not remove `unipi-id`? (I understand that `rtc_unipi` is required for RTC.)
>
> Can I just access DI via [...]

We do not endorse .NET solutions, nor we test it, so we can't provide any information about this. If it supports standard Linux IO, it should work. I am closing this issue, as it became off-topic, Unipi Technology does provide paid support/consultations for specific needs.

closed 2025-01-08 by cleveHEX