UnipiTechnology/evok #207: [Bug]: Invalid watchdog property name in hw definitions

cleveHEX, 2025-07-15:
### Prerequisites

- [x] I am running the latest Evok version
- [x] I have searched other [issues](https://github.com/UniPiTechnology/evok/issues)

### Describe the bug

Watchdog property in documentation is called `nv_save_coil`, but Evok uses `nv_sav_coil`

### Expected behavior

Documentation matches reality

### Actual behavior

Documentation does not match reality

miroslav-ondra, 2026-10-09:
Property is obsolete and will be removed

miroslav-ondra, 2026-10-09:
This will be resolved in Evok 3.1 by commit f298e87 (Reject the deprecated nv_save of a watchdog).

`nv_sav_coil` of the WD feature is not used any more, the docs of the HW definitions name it `nv_sav_coil` and describe it as not used. The settings of a unit are saved by the NV_SAVE feature (device `nv_save`), `nv_save` of a watchdog returns an error.

closed 2026-11-02 by matoushybl